raspiblitz/raspiblitz #1: SD-Image Buildscript

rootzoll, 2018-07-17:
At the moment the RaspiBlitz works with a prepared SD-Card image that can be downloaded. Its always a trust issue and the detail of this images lack of documentation. 

The goal for the future would be to create a shell script that contains all the changes and installs on then OS base image. This way for anybody not trusting the prepared SD-Card image could invest some more time, take a plain Rasbian https://www.raspberrypi.org/downloads/raspbian/ ,run the shell script on and build a SD-Image fresh.

pieman64, 2018-07-20:
I am downloading the image at the moment. Looks to be 2.1GB compressed, how big when it's expanded and is 8GB card OK (notice you show 16GB)?

rootzoll, 2018-07-20:
16GB card is recommended and almost same as cheap as 8GB. You could get the
image onto a 8GB .. but some cards are a little bit different in size. With
16GB card you are safe.

rootzoll, 2018-08-06:
build script is now  available - see: https://github.com/rootzoll/raspiblitz#build-the-sd-card-image

testing and feedback welcome

rootzoll, 2018-08-06:
still some bugs after testing:
- screen orientation durning 270° not effective yet
- cli autologin for user pi not working

rootzoll, 2018-08-08:
OK ... looks good so far on my test runs. So check https://github.com/rootzoll/raspiblitz#build-the-sd-card-image to build your own SD card image. Closing this issue until further bug reports.

closed 2018-08-08 by rootzoll

nyxnor, 2021-04-20:
Seeing this was the first ever issue of Raspiblitz (in this repo) is amazing, managing for zero trust since the beginning :100: :1st_place_medal:

commit dc32500d (2018-08-06): build sd card doc - close #1