raspiblitz/raspiblitz #4: Run RaspiBlitz behind TOR

rootzoll, 2018-07-19:
Running the RaspiBlitz behind Tor as a hidden service would be great ... not just because beeing more anon. Its perfect to make the node publicly available even when behind a NAT - without having to set port forwarding, etc.

See discussions for inspiration here: https://twitter.com/Stadicus3000/status/1017704449303867393

First step would be to write a shell script that installs and setups a TOR hidden service and takes care about the settings. If we then have also a script that would reverse this process, we could add a option into the main menu to switch TOR on and off.

rootzoll, 2018-07-23:
Also check: https://github.com/Sjors/armbian-bitcoin-core/commit/ef915416546d1dd0a41f8fc479a2a67f9ae8a8f8

rootzoll, 2018-08-15:
Getting the seed nodes fresh instead of having them static in `96addTorService.sh` would an improvement.

vnnkl, 2018-08-15:
im not familiar with dig yet, will look into it and leave this here for memo:
https://bitnodes.earn.com/api/#dns-seeder

rootzoll, 2018-08-18:
TOR support for Shango Wallet is coming:
https://github.com/neogeno/shango-lightning-wallet/issues/37
This would mean that if we can get LND running behind TOR .. it could connect without any dyndns or port-forwarding-configuration-on-router. This sounds like the easiest first way we should approach.

PR #39 (2018-08-14): WIP: add TorService script
scripted worked on my machine,
no way to reverse it yet, so adding only for now

- [ ] needs menu item

Let's discuss how much we want in there
closes #4
closed 2018-09-04 by rootzoll