raspiblitz/raspiblitz #9: Can't get past downloading the blockchain

pettergo, 2018-07-23:
I've gotten as far as choosing the first option to download the blockchain via torrent. When the unix starts running it keeps stopping at:

*** Moving Files ***
moving files ...
mv: cannot move '/mnt/hdd/raspiblitz-hdd-2018-07-16/blocks' to '/mnt/hdd/blocks': Permission denied
mv: cannot move '/mnt/hdd/raspiblitz-hdd-2018-07-16/chainstate' to '/mnt/hdd/chainstate': Permission denied
mv: cannot move '/mnt/hdd/raspiblitz-hdd-2018-07-16/testnet3' to '/mnt/hdd/testnet3': Permission denied
rm: descend into write-protected directory '/mnt/hdd/raspiblitz-hdd-2018-07-16'? 
--------------------------
Now, i'm not sure if its working and i don't see anything or if i'm supposed to do anything.  If I enter my password then I get:

*** Next Step  ***
You can now use this HDD as a source to copy the Blockchain during the setup of another RaspiBlitz.

*** Checking HDD ***
FAIL - HDD is empty.
admin@raspberrypi:~ $ 
----------------------------
In the mean time, I'm downloading the blockchain on a ssd from my computer in case I need to go with option 2. I'm working on MacOS Sierra and know next to nothing about Unix commands.  but i'm learning.

pettergo, 2018-07-24:
Maybe. Would that mean that I just need to keep trying? or is it the wrong torrent file?

I found this line:
> [2018-07-23 16:19:12.160] raspiblitz-hdd-2018-07-16: write failed for "raspiblitz-hdd-2018-07-16/blocks/index/114130.ldb": No space left on device

I wonder if its not accessing the hdd? But I don't know, this is my first time doing anything remotely like this.  I've almost completed downloading the blockchain onto a ssd and i'll try option 2 later today hopefully. that might work easier.

rootzoll, 2018-07-24:
Torrent file seems ok. Looks like something got messed up with the permission settings for the hdd - could be that the torrent process does not have the permission to write the files or that somehow the inode limit is reached. I will run a test and see if I can reproduce ... will keep you updated,

rootzoll, 2018-07-24:
OK first the "Permission denied" seems to happen because transmission runs with sudo and file owner is root … but the movement of file runs without sudo - seems like I packaged an old version of the file. Working on a fix.

gabrieldov, 2018-07-25:
Yes, please fix that - same issue here. Also there are not a lot of people seeding the torrent. Should I halt the process here and wait for the fix? Cheers

charlesrocket, 2018-07-25:
script ignored prepared hdd, formatting fully prepared hdd, still failing to sync with testnet or download from any seeder.

rootzoll, 2018-07-25:
@gabrieldov @pettergo @charlesrocket I pushed a first fix for the download problem. I am testing right now - looks good so far. It would be great if you can help on testing, before I package a new SD card image. Just simple cancel on main menu and replace all shell scripts on your raspiBlitz in folder home/admin with the latest shell scripts in this gitrepo (folder home.admin) ... run a chmod +x /home/admin/*.sh after replacement if needed

@charlesrocket about the preparedHHD - what format is the HDD (ext4, exfat, ntfs, fat32)? And can you screenshot the content of the root directory?

charlesrocket, 2018-07-25:
@rootzoll testing scripts. it was exfat with the blockchain inside, it got formatted so no screenshots) but osx not reading drives formatted by rpi and always want to reformat it, so maybe prepared on osx drives will not work

rootzoll, 2018-07-25:
I uploaded a new SD card image and linked it in the README.md: http://wiki.fulmo.org/downloads/raspiblitz-2018-07-25.img.gz

I integrated the memory settings for better torrent and also added a download check and a 3 times retry around the torrent download. My test is till running ... but I think the main issue is fixed. Please let me know your experiences.

But my torrent download is still quite slow ... so if anybody likes to help out with seeding that would be great: https://github.com/rootzoll/raspiblitz/raw/master/home.admin/assets/raspiblitz-hdd-2018-07-16.torrent

charlesrocket, 2018-07-27:
torrent file is in 99% state for hours, all last pieces are failed checksum

vnnkl, 2018-07-27:
did you try to re-check / restart?

gabrieldov, 2018-07-27:
Will try to download the blockchain from other source... Would be good to have multiple sources to choose from (even straight from Satoshi nodes)

rootzoll, 2018-07-27:
> Will try to download the blockchain from other source... Would be good to have multiple sources to choose from (even straight from Satoshi nodes)

The "SYNC" option is exactly to download thru the Bitcoin Nodes and
index the Blockchain yourself. For Testnet (which is the starting
default) this should even be done in a reasonable time. If you switch
to mainnet later, it wil try to sync mainnet ... that can be a very
long ride, sometime even up to a week or so.

Please remember that the RaspiBlitz started as a setup for on location
hackathons where you will get a prepared HDD or can copy from an
already synced HDD. The download version to start at home was still
very experimental, when the project hit the news on twitter. Thanks
for giving feedback on it. It looks like that the torrent-download of
the peindexed Blockchain is showing the speed hoped for - I am at
least not a torrent expert to find the problem. I am thinink to change
this maybe to plain old HTTP Download - at least for this early days
until someone can help out to get the torrents speed right (with
seeding and fine tuning).

charlesrocket, 2018-07-27:
this is on 99%
```
*** Checking Download ***
wait a moment
download size is(231230512)

FAIL - download is not 461446200
--> RETRY(1) in 10 secs
```

and no changes. sometimes see peers tho but nothing good downloads

rootzoll, 2018-07-27:
Ah OK ... sorry maybe my bad, I may have got the file size to check. Let me correct this quick ..

rootzoll, 2018-07-27:
@charlesrocket  I pushed the updated shell scripts ... please replace them and try to finish the downlaod again.

charlesrocket, 2018-07-27:
@rootzoll got it spinning!

rootzoll, 2018-07-27:
@charlesrocket pushed updated shell scripts, please try again

charlesrocket, 2018-07-27:
@rootzoll updated`raspiblitz-2018-07-25.img.gz`?

rootzoll, 2018-07-27:
SD card image is still in processing ... the RPC error locks strange. Lets hope this is a fallout from updating scripts in the middle and with fresh update this will fix. Otherwise I will check on this tomorrow.

rootzoll, 2018-07-27:
@charlesrocket README updated ... new image is: http://wiki.fulmo.org/downloads/raspiblitz-2018-07-27c.img.gz

charlesrocket, 2018-07-27:
@rootzoll awesome, thanks! testing right now

rivelli87, 2018-07-27:
My issue is pretty similar. Downloading the torrent seemed fine in the beginning. Then my HDD for some reason changes to read only, forcing the script. Amusingly, I could sudo mount /mnt/hdd and restart the download script. It works sometimes for a few minutes or more...or less. I have no idea why. The first time took a few hours and got me to 9.8%.

//NORMAL PART
[2018-07-27 19:21:11.620] raspiblitz-hdd-2018-07-16: Piece 98794, which was just downloaded, failed its checksum test
[2018-07-27 19:21:11.884] raspiblitz-hdd-2018-07-16: Piece 110364, which was just downloaded, failed its checksum test
[2018-07-27 19:21:12.061] raspiblitz-hdd-2018-07-16: Piece 105276, which was just downloaded, failed its checksum test
Progress: 10.2%, dl from 0 of 2 peers (2.60 M[2018-07-27 19:21:17.023] raspiblitz-hdd-2018-07-16: Piece 105815, which was just downloaded, failed its checksum test
[2018-07-27 19:21:18.415] raspiblitz-hdd-2018-07-16: Piece 106257, which was just downloaded, failed its checksum test
//PART WHERE I SEE IT'S ABOUT TO QUIT ON ME
Progress: 10.2%,[2018-07-27 19:21:20.046] raspiblitz-hdd-2018-07-16: write failed for "raspiblitz-hdd-2018-07-16/blocks/index/138638.ldb": Read-only file system
[2018-07-27 19:21:20.046] raspiblitz-hdd-2018-07-16: Read-only file system (/mnt/hdd/raspiblitz-hdd-2018-07-16/blocks/index/138638.ldb)

Error:: Read-only file system (/mnt/hdd/raspiblitz-hdd-2018-07-16/blocks/index/138638.ldb)
[2018-07-27 19:21:20.320] raspiblitz-hdd-2018-07-16: Pausing
[2018-07-27 19:21:20.635] Saved "/root/.config/transmission/settings.json"
Progress: 10.2%, dl from 2 of 2 peers (1.97 MB/s), ul to 0 (0 kB/s) [0.00]
[2018-07-27 19:21:22.906] Couldn't open "/mnt/hdd/raspiblitz-hdd-2018-07-16/blocks/rev01208.dat.part": Input/output error
[2018-07-27 19:21:22.907] raspiblitz-hdd-2018-07-16: tr_fdFileCheckout failed for "/mnt/hdd/raspiblitz-hdd-2018-07-16/blocks/rev01208.dat.part": Input/output error
[2018-07-27 19:21:24.101] Saved "/root/.config/transmission/resume/raspiblitz-hdd-2018-07-16.389af3fc8216a244.resume"
[2018-07-27 19:21:24.109] DHT: Not saving nodes, DHT not ready
[2018-07-27 19:21:24.109] Port Forwarding: Stopped
[2018-07-27 19:21:24.858] Port Forwarding (UPnP): Stopping port forwarding through "http://10.0.0.1:49152/upnp/control/WANIPConnection0", service "urn:schemas-upnp-org:service:WANIPConnection:1"
[2018-07-27 19:21:24.858] Port Forwarding: State changed from "Forwarded" to "Not forwarded"
[2018-07-27 19:21:24.858] raspiblitz-hdd-2018-07-16: Removing torrent
[2018-07-27 19:21:24.858] raspiblitz-hdd-2018-07-16: Pausing
[2018-07-27 19:21:24.861] Couldn't open "/mnt/hdd/raspiblitz-hdd-2018-07-16/chainstate/265419.ldb.part": Input/output error
[2018-07-27 19:21:24.862] raspiblitz-hdd-2018-07-16: tr_fdFileCheckout failed for "/mnt/hdd/raspiblitz-hdd-2018-07-16/chainstate/265419.ldb.part": Input/output error
[2018-07-27 19:21:26.066] Saved "/root/.config/transmission/resume/raspiblitz-hdd-2018-07-16.389af3fc8216a244.resume"
[2018-07-27 19:21:26.084] Saved "/root/.config/transmission/stats.json"

*** Moving Files ***
moving files ...
mv: cannot stat '/mnt/hdd/raspiblitz-hdd-2018-07-16/*': Input/output error
rm: cannot remove '/mnt/hdd/raspiblitz-hdd-2018-07-16': Read-only file system

*** Next Step  ***
You can now use this HDD as a source to copy the Blockchain during the setup of another RaspiBlitz.

*** Checking HDD ***
FAIL - HDD is empty.

MichaelCb1, 2018-07-28:
Mine is also alike:

*** Downloading HDD ***
transmission-cli 2.92 (14714)
[2018-07-28 15:52:59.195] Transmission 2.92 (14714) started
[2018-07-28 15:52:59.196] RPC Server: Adding address to whitelist: 127.0.0.1
[2018-07-28 15:52:59.197] DHT: Generating new id
[2018-07-28 15:52:59.528] Port Forwarding (NAT-PMP): initnatpmp succeeded (0)
[2018-07-28 15:52:59.528] Port Forwarding (NAT-PMP): sendpublicaddressrequest succeeded (2)
[2018-07-28 15:52:59.873] Saved "/root/.config/transmission/torrents/raspiblitz-hdd-2018-07-16.389af3fc8216a244.torrent"
Progress: 0.0%, dl from 0 of 0 peers (0 kB/s), ul t[2018-07-28 15:53:02.617] Port Forwarding (UPnP): Found Internet Gateway Device "http://192.168.0.1:5000/ctl/IPConn"
[2018-07-28 15:53:02.618] Port Forwarding (UPnP): Local Address is "192.168.0.200"
[2018-07-28 15:53:02.912] Port Forwarding (UPnP): Port forwarding through "http://192.168.0.1:5000/ctl/IPConn", service "urn:schemas-upnp-org:service:WANIPConnection:1". (local address: 192.168.0.200:51413)
[2018-07-28 15:53:02.912] Port Forwarding (UPnP): Port forwarding successful!
[2018-07-28 15:53:02.912] Port Forwarding: State changed from "Not forwarded" to "Forwarded"
[2018-07-28 15:53:02.912] raspiblitz-hdd-2018-07-16: Pausing
[2018-07-28 15:53:02.912] Changed open file limit from 128000 to 1024
[2018-07-28 15:53:02.949] Saved "/root/.config/transmission/settings.json"
Progress: 0.0%, dl from 0 of 0 peers (0 kB/s), ul to 0 (0 kB/s) [None]          
[2018-07-28 15:53:03.334] Saved "/root/.config/transmission/resume/raspiblitz-hdd-2018-07-16.389af3fc8216a244.resume"
[2018-07-28 15:53:03.354] raspiblitz-hdd-2018-07-16: Queued for verification
[2018-07-28 15:53:03.502] raspiblitz-hdd-2018-07-16: Verifying torrent
[2018-07-28 15:53:03.504] DHT: Not saving nodes, DHT not ready
[2018-07-28 15:53:03.504] Port Forwarding: Stopped
[2018-07-28 15:53:03.688] Port Forwarding (UPnP): Stopping port forwarding through "http://192.168.0.1:5000/ctl/IPConn", service "urn:schemas-upnp-org:service:WANIPConnection:1"
[2018-07-28 15:53:03.688] Port Forwarding: State changed from "Forwarded" to "Starting"
[2018-07-28 15:53:03.688] raspiblitz-hdd-2018-07-16: Removing torrent
[2018-07-28 15:53:03.688] raspiblitz-hdd-2018-07-16: Pausing
[2018-07-28 15:53:04.080] Saved "/root/.config/transmission/resume/raspiblitz-hdd-2018-07-16.389af3fc8216a244.resume"

*** Checking Download ***
wait a moment
du: cannot access '/mnt/hdd/raspiblitz-hdd-2018-07-16/': No such file or directory
download size is(0)

FAIL - download is not 230206512
--> RETRY(1) in 10 secs

*** Downloading HDD ***
transmission-cli 2.92 (14714)
[2018-07-28 15:53:19.536] Transmission 2.92 (14714) started
[2018-07-28 15:53:19.538] RPC Server: Adding address to whitelist: 127.0.0.1
[2018-07-28 15:53:19.539] DHT: Generating new id
[2018-07-28 15:53:19.869] Port Forwarding (NAT-PMP): initnatpmp succeeded (0)
[2018-07-28 15:53:19.869] Port Forwarding (NAT-PMP): sendpublicaddressrequest succeeded (2)
[2018-07-28 15:53:20.300] Saved "/root/.config/transmission/torrents/raspiblitz-hdd-2018-07-16.389af3fc8216a244.torrent"
[2018-07-28 15:53:22.345] Port Forwarding (UPnP): Found Internet Gateway Device "http://192.168.0.1:5000/ctl/IPConn"
[2018-07-28 15:53:22.346] Port Forwarding (UPnP): Local Address is "192.168.0.200"
[2018-07-28 15:53:22.568] Port Forwarding (UPnP): Port forwarding through "http://192.168.0.1:5000/ctl/IPConn", service "urn:schemas-upnp-org:service:WANIPConnection:1". (local address: 192.168.0.200:51413)
[2018-07-28 15:53:22.568] Port Forwarding (UPnP): Port forwarding successful!
[2018-07-28 15:53:22.568] Port Forwarding: State changed from "Not forwarded" to "Forwarded"
Progress: 0.0%, dl from 0 of 0 peers (0 kB/s), ul t[2018-07-28 15:53:25.730] Changed open file limit from 128000 to 1024
[2018-07-28 15:53:25.775] Couldn't preallocate file "/mnt/hdd/raspiblitz-hdd-2018-07-16/blocks/blk00677.dat.part" (sparse, size: 134138065): No space left on device
[2018-07-28 15:53:25.775] raspiblitz-hdd-2018-07-16: tr_fdFileCheckout failed for "/mnt/hdd/raspiblitz-hdd-2018-07-16/blocks/blk00677.dat.part": No space left on device
[2018-07-28 15:53:25.775] raspiblitz-hdd-2018-07-16: No space left on device (/mnt/hdd/raspiblitz-hdd-2018-07-16/blocks/blk00677.dat)
Progress: 0.0%, dl from 0 of 0 peers (1.27 MB/s), ul to 0 (0 kB/s) [0.00]       
Error:: No space left on device (/mnt/hdd/raspiblitz-hdd-2018-07-16/blocks/blk00677.dat)
[2018-07-28 15:53:26.211] raspiblitz-hdd-2018-07-16: Pausing
[2018-07-28 15:53:26.213] Couldn't preallocate file "/mnt/hdd/raspiblitz-hdd-2018-07-16/testnet3/blocks/blk00074.dat.part" (sparse, size: 134077880): No space left on device
[2018-07-28 15:53:26.213] raspiblitz-hdd-2018-07-16: tr_fdFileCheckout failed for "/mnt/hdd/raspiblitz-hdd-2018-07-16/testnet3/blocks/blk00074.dat.part": No space left on device
[2018-07-28 15:53:26.365] Saved "/root/.config/transmission/settings.json"
Progress: 0.0%, dl from 0 of 1 peers (2.53 MB/s), ul to 0 (0 kB/s) [0.00]       
[2018-07-28 15:53:26.505] Saved "/root/.config/transmission/resume/raspiblitz-hdd-2018-07-16.389af3fc8216a244.resume"
[2018-07-28 15:53:26.510] DHT: Not saving nodes, DHT not ready
[2018-07-28 15:53:26.510] Port Forwarding: Stopped
[2018-07-28 15:53:26.942] Port Forwarding (UPnP): Stopping port forwarding through "http://192.168.0.1:5000/ctl/IPConn", service "urn:schemas-upnp-org:service:WANIPConnection:1"
[... cut ...]
[2018-07-28 15:53:50.754] raspiblitz-hdd-2018-07-16: Pausing
[2018-07-28 15:53:50.755] raspiblitz-hdd-2018-07-16: write failed for "raspiblitz-hdd-2018-07-16/blocks/blk00915.dat": No space left on device
[2018-07-28 15:53:51.101] Saved "/root/.config/transmission/resume/raspiblitz-hdd-2018-07-16.389af3fc8216a244.resume"
[2018-07-28 15:53:51.121] Saved "/root/.config/transmission/stats.json"

*** Checking Download ***
wait a moment
download size is(188589)

FAIL - download is not 228158512
All Retry FAILED

rootzoll, 2018-07-29:
@vnnkl @MichaelCb1 @rivelli87 @gabrieldov @pettergo @charlesrocket 
I created a new torrent and packaged it into a new sdcard image - see updated link in README.me

Any help on seeding the new two torrents is welcome:
http://wiki.fulmo.org/downloads/raspiblitz-litecoin-2018-07-29.torrent
http://wiki.fulmo.org/downloads/raspiblitz-bitcoin-2018-07-16.torrent

Jut in case the torrent download has still any problems I added a new FTP download option (hosting from home - so be gentle). If anybody can spend a 250GB FTP space (with anon/read-only account) on a good flat-traffic server let me know.

pettergo, 2018-07-30:
Been afk for this weekend with the family.  Just tried raspiblitz-2018-07-29b.img but it's getting stuck still.  
----------------------------
 
torrentFile: raspiblitz-bitcoin-2018-07-16

*** Downloading TORRENT ***
IN CASE DOWNLOAD DOES NOT START OR TOO SLOW:
CTRL+z start ./10setupBlitz.sh choose other option
***************************

transmission-cli 2.92 (14714)
[2018-07-30 11:09:44.457] Transmission 2.92 (14714) started
[2018-07-30 11:09:44.458] RPC Server: Adding address to whitelist: 127.0.0.1
[2018-07-30 11:09:44.459] DHT: Generating new id
[2018-07-30 11:09:44.790] Port Forwarding (NAT-PMP): initnatpmp succeeded (0)
[2018-07-30 11:09:44.790] Port Forwarding (NAT-PMP): sendpublicaddressrequest succeeded (2)
[2018-07-30 11:09:45.145] Saved "/root/.config/transmission/torrents/raspiblitz-bitcoin-2018-07-16.5bfbb354685eec9a.torrent"
[2018-07-30 11:09:46.813] Port Forwarding (UPnP): Found Internet Gateway Device "http://192.168.178.1:49000/igdupnp/control/WANIPConn1"
[2018-07-30 11:09:46.813] Port Forwarding (UPnP): Local Address is "192.168.178.20"
[2018-07-30 11:09:46.825] Port Forwarding (UPnP): Port forwarding through "http://192.168.178.1:49000/igdupnp/control/WANIPConn1", service "urn:schemas-upnp-org:service:WANIPConnection:1". (local address: 192.168.178.20:51413)
[2018-07-30 11:09:46.825] Port Forwarding: State changed from "Not forwarded" to "Starting"
[2018-07-30 11:09:46.825] raspiblitz-bitcoin-2018-07-16: Pausing
[2018-07-30 11:09:46.825] Changed open file limit from 128000 to 1024
[2018-07-30 11:09:47.023] Saved "/root/.config/transmission/settings.json"
Progress: 0.0%, dl from 0 of 0 peers (0 kB/s), ul to 0 (0 kB/s) [None]          
[2018-07-30 11:09:47.258] Saved "/root/.config/transmission/resume/raspiblitz-bitcoin-2018-07-16.5bfbb354685eec9a.resume"
[2018-07-30 11:09:47.276] raspiblitz-bitcoin-2018-07-16: Queued for verification
[2018-07-30 11:09:47.424] raspiblitz-bitcoin-2018-07-16: Verifying torrent
[2018-07-30 11:09:47.425] DHT: Not saving nodes, DHT not ready
[2018-07-30 11:09:47.425] Port Forwarding: Stopped
[2018-07-30 11:09:47.425] raspiblitz-bitcoin-2018-07-16: Removing torrent
[2018-07-30 11:09:47.425] raspiblitz-bitcoin-2018-07-16: Pausing
[2018-07-30 11:09:47.893] Saved "/root/.config/transmission/resume/raspiblitz-bitcoin-2018-07-16.5bfbb354685eec9a.resume"
OK - Download closed

*** Checking TORRENT ***
wait a moment
du: cannot access '/mnt/hdd/raspiblitz-bitcoin-2018-07-16/': No such file or directory
download size is(0) needs to be minimum(-1024000)
*** Moving Files ***
moving files ...
mv: cannot stat '/mnt/hdd/raspiblitz-bitcoin-2018-07-16': No such file or directory

*** Checking HDD ***
FAIL - HDD is empty.
admin@raspberrypi:~ $ 
------------------------------
Selecting option for Download doesn't seem to work either:  

network(bitcoin)
list()
url(ftp://anonymous:anonymous@tll9xsfkjht8j26z.myfritz.net/raspiblitz-bitcoin-2018-07-16)
size(100)
targetPath(tll9xsfkjht8j26z.myfritz.net/raspiblitz-bitcoin-2018-07-16)

*** Downloading HDD / FTP ***
OK

*** Checking Download ***
du: cannot access '/mnt/hdd/tll9xsfkjht8j26z.myfritz.net/raspiblitz-bitcoin-2018-07-16': No such file or directory
download size is(0) needs to be minimum(100})

rootzoll, 2018-07-30:
@pettergo this is very strange - can anybody give me a hint why this is happening?

50torrentHDD.sh ) transmission is showing no errors, but is acting like it downloaded the torrent data in no time and is proceeding.

50downloadHDD.sh ) wget is writing no errors, no outout at all and the script continues.

Can anybody report the same issue?

pettergo, 2018-07-31:
are there any workspaces in berlin that i can bring my raspiblitz to and get some help? i know there was a workshop there earlier this summer.

rootzoll, 2018-07-31:
We can give support for raspiblitz on every #lightninghack day in Berlin .. next one will be around the weekend of the 1st of september - follow on twitter or ask @fulmolightning for details

charlesrocket, 2018-08-05:
@rootzoll had same issue with old torrent file, errors on pre located mem, reboot asked for maintenance - so looked like its better to redownload trnt and reflash SD. can't reproduce with new image

pettergo, 2018-08-07:
torrentFile: raspiblitz-bitcoin-2018-07-16

*** Downloading TORRENT ***
IN CASE DOWNLOAD DOES NOT START OR TOO SLOW:
CTRL+z start ./10setupBlitz.sh choose other option
***************************

transmission-cli 2.92 (14714)
[2018-08-07 14:35:01.619] Transmission 2.92 (14714) started
[2018-08-07 14:35:01.620] RPC Server: Adding address to whitelist: 127.0.0.1
[2018-08-07 14:35:01.621] DHT: Generating new id
[2018-08-07 14:35:01.951] Port Forwarding (NAT-PMP): initnatpmp succeeded (0)
[2018-08-07 14:35:01.951] Port Forwarding (NAT-PMP): sendpublicaddressrequest succeeded (2)
[2018-08-07 14:35:02.381] Saved "/root/.config/transmission/torrents/raspiblitz-bitcoin-2018-07-16.5bfbb354685eec9a.torrent"
[2018-08-07 14:35:04.022] Port Forwarding (UPnP): Found Internet Gateway Device "http://192.168.178.1:49000/igdupnp/control/WANIPConn1"
[2018-08-07 14:35:04.022] Port Forwarding (UPnP): Local Address is "192.168.178.20"
[2018-08-07 14:35:04.035] Port Forwarding (UPnP): Port forwarding through "http://192.168.178.1:49000/igdupnp/control/WANIPConn1", service "urn:schemas-upnp-org:service:WANIPConnection:1". (local address: 192.168.178.20:51413)
[2018-08-07 14:35:04.035] Port Forwarding: State changed from "Not forwarded" to "Starting"
Progress: 0.0%, dl fro[2018-08-07 14:35:09.039] Port Forwarding: State changed from "Starting" to "???"
Progress: 0.0%, dl from 3 of 3 peers (99 kB/s), ul to 0 (0 kB/s) [0.[2018-08-07 14:35:19.022] Changed open file limit from 128000 to 1024
[2018-08-07 14:35:19.032] Couldn't create "/mnt/hdd/raspiblitz-bitcoin-2018-07-16": No space left on device
[2018-08-07 14:35:19.032] Couldn't create "/mnt/hdd/raspiblitz-bitcoin-2018-07-16/blocks/index": No space left on device
[2018-08-07 14:35:19.032] raspiblitz-bitcoin-2018-07-16: tr_fdFileCheckout failed for "/mnt/hdd/raspiblitz-bitcoin-2018-07-16/blocks/index/145195.ldb.part": No space left on device
[2018-08-07 14:35:19.032] raspiblitz-bitcoin-2018-07-16: No space left on device (/mnt/hdd/raspiblitz-bitcoin-2018-07-16/blocks/index/145195.ldb)

Error:: No space left on device (/mnt/hdd/raspiblitz-bitcoin-2018-07-16/blocks/index/145195.ldb)
[2018-08-07 14:35:19.344] raspiblitz-bitcoin-2018-07-16: Pausing
[2018-08-07 14:35:19.472] Saved "/root/.config/transmission/settings.json"
Progress: 0.0%, dl from 3 of 3 peers (307 kB/s), ul to 0 (0 kB/s) [0.00]        
[2018-08-07 14:35:22.498] Couldn't create "/mnt/hdd/raspiblitz-bitcoin-2018-07-16": No space left on device
[2018-08-07 14:35:22.498] Couldn't create "/mnt/hdd/raspiblitz-bitcoin-2018-07-16/blocks/index": No space left on device
[2018-08-07 14:35:22.498] raspiblitz-bitcoin-2018-07-16: tr_fdFileCheckout failed for "/mnt/hdd/raspiblitz-bitcoin-2018-07-16/blocks/index/115521.ldb.part": No space left on device
[2018-08-07 14:35:22.888] Saved "/root/.config/transmission/resume/raspiblitz-bitcoin-2018-07-16.5bfbb354685eec9a.resume"
[2018-08-07 14:35:22.906] DHT: Not saving nodes, DHT not ready
[2018-08-07 14:35:22.906] Port Forwarding: Stopped
[2018-08-07 14:35:22.906] raspiblitz-bitcoin-2018-07-16: Removing torrent
[2018-08-07 14:35:22.906] raspiblitz-bitcoin-2018-07-16: Pausing
[2018-08-07 14:35:22.906] Couldn't create "/mnt/hdd/raspiblitz-bitcoin-2018-07-16": No space left on device
[2018-08-07 14:35:22.906] Couldn't create "/mnt/hdd/raspiblitz-bitcoin-2018-07-16/blocks/index": No space left on device
[2018-08-07 14:35:22.906] raspiblitz-bitcoin-2018-07-16: tr_fdFileCheckout failed for "/mnt/hdd/raspiblitz-bitcoin-2018-07-16/blocks/index/090452.ldb.part": No space left on device
[2018-08-07 14:35:22.912] Saved "/root/.config/transmission/stats.json"
[2018-08-07 14:35:22.929] DHT: Attempting bootstrap from dht.transmissionbt.com
OK - Download closed

*** Checking TORRENT ***
wait a moment
du: cannot access '/mnt/hdd/raspiblitz-bitcoin-2018-07-16/': No such file or directory
download size is(0) needs to be minimum(-1024000)
*** Moving Files ***
moving files ...
mv: cannot stat '/mnt/hdd/raspiblitz-bitcoin-2018-07-16': No such file or directory

*** Checking HDD ***
FAIL - HDD is empty.
admin@raspberrypi:~ $ 

------------------------------
What would you do?

pettergo, 2018-08-07:
network(bitcoin)
list()
url(ftp://anonymous:anonymous@tll9xsfkjht8j26z.myfritz.net/raspiblitz-bitcoin-2018-07-16)
size(100)
targetPath(tll9xsfkjht8j26z.myfritz.net/raspiblitz-bitcoin-2018-07-16)

*** Downloading HDD / FTP ***
OK

*** Checking Download ***
du: cannot access '/mnt/hdd/tll9xsfkjht8j26z.myfritz.net/raspiblitz-bitcoin-2018-07-16': No such file or directory
download size is(0) needs to be minimum(100})
-----------------------------------------
Just let me know if I'm the only one with this problem and I'll shut up. Otherwise I'll keep trying the new images you upload and give feedback.

charlesrocket, 2018-08-07:
@pettergo try to wipe hdd and reflash with last image, looks like you still have old torrent files cached. i would do a new clean run

pettergo, 2018-08-07:
AHA! Thanks @charlesrocket, it's downloading from the sever as I type. 
I believe that I can set this thread to rest.

closed 2018-08-07 by pettergo