raspiblitz/raspiblitz #17: undervoltage detected

charlesrocket, 2018-07-27:
LCD screen throw the error before the reboot `undervoltage detected` during LN setup, with following hdd errors. trying to recreate

charlesrocket, 2018-07-27:
changing power source giving undervoltage on boot right away. too much load on hdd during init or local power source

rootzoll, 2018-07-27:
Do you have the Power Supply from the Shopping list?
https://www.amazon.de/dp/B01E75SB2C

To run an external HDD the Pi needs some extra power - with the power supply from the shopping list I never had any running it with one HDD.

charlesrocket, 2018-07-27:
@rootzoll im using different one from US list, but it doesn't really look like lack of power, at this time. it would throw the error on boot. right now im having endless cycle of
```0.00%
error code: -28
error message:
Loading block index...
(standard_in) 1: syntax error
./70initLND.sh: 24: [: -eq: unexpected operator
./70initLND.sh: 27: [: -eq: unexpected operator
waiting to start sync
```
```
0.00%
error code: -28
error message:
Rewinding blocks...
(standard_in) 1: syntax error
./70initLND.sh: 24: [: -eq: unexpected operator
./70initLND.sh: 27: [: -eq: unexpected operator
waiting to start sync
```
```
0.00%
error code: -28
error message:
Verifying blocks...
(standard_in) 1: syntax error
./70initLND.sh: 24: [: -eq: unexpected operator
./70initLND.sh: 27: [: -eq: unexpected operator
waiting to start sync
```

rootzoll, 2018-07-28:
I updated the US Shopping list, because the power supply there had just 2.5A - I am running one with 3.0A .. the extra 0.5A seems to protect you on power peaks from the HDD.

About the looping errors on sync - see https://github.com/rootzoll/raspiblitz/issues/18

charlesrocket, 2018-07-28:
i was afraid that its broken torrent file, and looks like it
```
2018-07-28 13:31:40 msghand thread start
2018-07-28 13:31:40 LevelDB read failure: Corruption: block checksum mismatch: /home/bitcoin/.bitcoin/testnet3/chainstate/024914.ldb
2018-07-28 13:31:40 Corruption: block checksum mismatch: /home/bitcoin/.bitcoin/testnet3/chainstate/024914.ldb
2018-07-28 13:31:40 Error: Error reading from database, shutting down.
2018-07-28 13:31:40 Error reading from database: Database corrupted
```

commit d363a177 (2018-07-29): close #17
closed 2018-07-29 by rootzoll