raspiblitz/raspiblitz #18: can't start bitcoind

charlesrocket, 2018-07-27:
```
waiting to start sync
0.00%
error code: -28
error message:
Loading block index...
(standard_in) 1: syntax error
./70initLND.sh: 24: [: -eq: unexpected operator
./70initLND.sh: 27: [: -eq: unexpected operator
waiting to start sync
0.00%
error code: -28
error message:
Loading block index...
(standard_in) 1: syntax error
./70initLND.sh: 24: [: -eq: unexpected operator
./70initLND.sh: 27: [: -eq: unexpected operator
waiting to start sync
0.00%
error code: -28
error message:
Rewinding blocks...
(standard_in) 1: syntax error
./70initLND.sh: 24: [: -eq: unexpected operator
./70initLND.sh: 27: [: -eq: unexpected operator
waiting to start sync
0.00%
```
then
```
./70initLND.sh: 27: [: -eq: unexpected operator
waiting to start sync
0.00%
error: couldn't connect to server: EOF reached (code 1)
(make sure server is running and you are connecting to the correct RPC port)
(standard_in) 1: syntax error
./70initLND.sh: 24: [: -eq: unexpected operator
./70initLND.sh: 27: [: -eq: unexpected operator
waiting to start sync
0.00%
error: couldn't connect to server: unknown (code -1)
(make sure server is running and you are connecting to the correct RPC port)
(standard_in) 1: syntax error
./70initLND.sh: 24: [: -eq: unexpected operator
./70initLND.sh: 27: [: -eq: unexpected operator
waiting to start sync
0.00%
error: couldn't connect to server: unknown (code -1)
(make sure server is 
```

rootzoll, 2018-07-28:
In 70initLND.sh before getting into the sync loop it should check if bitcoind is running by:

`systemctl status bitcoind.service | grep -c running` > 0

Can you check manual what you get there - maybe we need to use another way to detect if bitcoind is running correctly - making an bitcoin-cli could be a better way.

I will run test today and see if I can reproduce this.

charlesrocket, 2018-07-28:
@rootzoll right after the booting:
```
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2018-07-28 15:39:11 CEST; 38s ago
  Process: 860 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitco
 Main PID: 861 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─861 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid
lines 1-7/7 (END)
```

rootzoll, 2018-07-29:
@charlesrocket I changed the is-bitcoin-running detection .. also added litecoin option on install. And just in case if the new torrent is still bad ... there is a FTP download option now. Check the README.md for the link to the new image.

closed 2018-07-29 by rootzoll