raspiblitz/raspiblitz #19: Enough power supply for HDD and RaspiBlitz

rocky3xr, 2018-07-28:
Hello, 
I'm ready to buy my amazon chart, just wondering if it is enough power the adapter to supply both the raspberry and the HDD

rootzoll, 2018-07-28:
I am running an extra streong power supply with 3A and have no problem runnin g the pi and one HDD anymore .. I made a change the USA shopping list to make sure that on all shopping lists have a power supply with at least 3A. Please check upadted lists before ordering.

rocky3xr, 2018-07-28:
Sure! So that means you run the entire setup with 2x 5v 3A that's why the Y cable, isn't it?

rootzoll, 2018-07-28:
The Y cable is just needed if you want to add a 2nd HDD. The 2nd HDD is just required if you want to sync the preindexed blockchain from another RaspiBlitz HDD during setup. 2nd HDD is not needed for normal running mode or if you choose another method to get the blockchain like download or sync.

In normal mode zu run the RaspiBlitz with one HDD and **one Power Suppply >=3A**

closed 2018-07-28 by rocky3xr