raspiblitz/raspiblitz #22: Piece 102422, which was just downloaded, failed its checksum test

goodc0re, 2018-07-29:
While downloading the chain data for raspiblitz via torrent, throughout the entire process starting from 1% or so, I saw the log being full of:
 
**"Piece 102422, which was just downloaded, failed its checksum test"** all the way through 99.9% 

[2018-07-29 11:25:49.378] raspiblitz-hdd-2018-07-16: Piece 102422, which was just downloaded, failed its checksum test
[2018-07-29 11:25:49.849] raspiblitz-hdd-2018-07-16: Piece 102422, which was just downloaded, failed its checksum test
[2018-07-29 11:25:50.517] raspiblitz-hdd-2018-07-16: Piece 102422, which was just downloaded, failed its checksum test
[2018-07-29 11:25:51.161] raspiblitz-hdd-2018-07-16: Piece 102422, which was just downloaded, failed its checksum test
[2018-07-29 11:25:51.605] raspiblitz-hdd-2018-07-16: Piece 102422, which was just downloaded, failed its checksum test
Progress: 99.9%, dl from 1 of 4 peers (3.11 MB/s), [2018-07-29 11:25:52.500] raspiblitz-hdd-2018-07-16: Piece 102422, which was just downloaded, failed its checksum test
[2018-07-29 11:25:53.136] raspiblitz-hdd-2018-07-16: Piece 102422, which was just downloaded, failed its checksum test
[2018-07-29 11:25:53.622] raspiblitz-hdd-2018-07-16: Piece 102422, which was just downloaded, failed its checksum test
[2018-07-29 11:25:54.032] raspiblitz-hdd-2018-07-16: Piece 102422, which was just downloaded, failed its checksum test
[2018-07-29 11:25:54.475] raspiblitz-hdd-2018-07-16: Piece 102422, which was just downloaded, failed its checksum test
[2018-07-29 11:25:55.196] raspiblitz-hdd-2018-07-16: Piece 102422, which was just downloaded, failed its checksum test
[2018-07-29 11:25:55.821] raspiblitz-hdd-2018-07-16: Piece 102422, which was just downloaded, failed its checksum test

Segmentation fault

*** Checking Download ***
wait a moment
download size is(231230516)

FAIL - download is not 460422200
--> RETRY(2) in 10 secs

kccheung, 2018-07-29:
+1 I got the same problem as well.

rootzoll, 2018-07-29:
@goodc0re @kccheung I created a new torrent and packaged it into a ne sdcard image - see link in readme.

Any help on seeding the new two torrents is welcome:
http://wiki.fulmo.org/downloads/raspiblitz-litecoin-2018-07-29.torrent
http://wiki.fulmo.org/downloads/raspiblitz-bitcoin-2018-07-16.torrent

Jut in case the torrent download has any problems I added a new FTP download option (hosting from home).

rootzoll, 2018-07-29:
I keep tissue open for feedback. Let me know if you have any further problems with the download with the new image ... this is something thats important to get right. Thanks for trying out his early stage prototype.

goodc0re, 2018-07-30:
I was using the raspiblitz setup to download the torrent, not the second HDD option. 

If I download the torrent I'd need a second HDD and Y-Cable, correct? 

Can I modify one of the scripts to change the torrent?

rootzoll, 2018-07-30:
> but it still is having problems with the block chain both with the torrent and the download option.

What exactly is not working? Please let me know details.

> Sorry i can't help seeding the torrent on my laptop (not enough space). can it seed from the raspiblitz? I'd be happy if it seeded from there. 

Let the raspiblitz do seeding after download of the torrent is a good idea. I open another issue for that.

charlesrocket, 2018-08-05:
download is good know, think this could be closed

closed 2018-08-05 by rootzoll