raspiblitz/raspiblitz #23: Litecoin Integration

rootzoll, 2018-07-29:
Today I uploaded the basic Litecoin integration. You can now choose in the first setup screen if you want to run your RaspiBlitz on Bitcoin or on Litecoin.

[image]

Im still running the setup process tests. Any help on trying out Litecoin on the Lightning Network with RaspiBlitz is welcome. Here are some links to try out:

https://github.com/ecurrencyhodler/Litecoin-Resources/blob/master/Lightning%20Network%20Apps.md

Let me know your feedback and issues.

vnnkl, 2018-07-29:
fyi, might be interesting later on
https://github.com/lightningnetwork/lnd/issues/1331

rootzoll, 2018-07-29:
@vnnkl thats sweet :) If you see any progress on this that we can utilize for the RaspiBlitz let me know. I also try to keep an ear open.

rootzoll, 2018-08-06:
seems everybody is happy with the litecoin integration :) .. I close issue for now

closed 2018-08-06 by rootzoll