raspiblitz/raspiblitz #24: Torrentseeding with RaspiBlitz

rootzoll, 2018-07-30:
To increase the number of seeders of the torrent file and to speed things up it would make sense to let the RaspiBlitz seed the torrent once it is downloaded.

There are some things to consider: At the moment after torrent download, the torrent gets renamed/moved and the directory is used for the bitcoind - so seeding an active used directory is not a good idea. To instead of moving the torrent it would need a copy after download. The 1TB HDD has enough space for that. But the copy could add another 2h of setup time.

The the best way would be from my perspektive: When user chooses torrent download, another dialoge is asking if the user would like to share/seed the torrent after download permantly. If "YES" then instead of movng the files they get copied after download. And Transmission gets added to the systemd services - starting everytime on boot.

openoms, 2019-02-18:
Transmission could be part of the Services menu, so one could help to seed even if did not use the torrent option to download the blockchain. Also could help to seed SDcard images, or anything really given enough disk space.

rootzoll, 2019-02-19:
Its a logical thing to do - and there is even another benefit of seeding new torrents: the user always has a fresh torrent copy of the blockchain available - so in case of data corruption no long waiting time, just copy over done :)

pxsocs, 2019-02-26:
Is there a way to limit the upload speed of the seed? 
I have an internet connection with an upload speed significantly lower than download speed. When the upload speed limit is hit, the download speeds drop significantly. 
On typical torrent download clients, there is usually an option to limit upload speed. Is there a config file where I could set that up?

geco91, 2019-02-26:
recommend to add this to v1.1 milestone ... because having a seeding blockchain ready means also that in case of data corruption of the one in action it means to have a replacement ready

rootzoll, 2019-04-11:
Decided to add an experimental "Backup Torrent Hosting" feature to the v1.2 release, that can be activated in the SERVICES section.

rootzoll, 2019-04-11:
TODO: make rtorrent NICE to reduce CPU priority
https://www.nixtutor.com/linux/changing-priority-on-linux-processes/

TODO: Open Incomming Ports on FireWall and set Port-Range on starting rtorrent
https://github.com/rakshasa/rtorrent/wiki/CLI-Usage

rootzoll, 2019-04-11:
OK now rtorrent plays NICE and just on IDLE. Ports or open on new sd card build and user info is there.

TODO: Test old torrent download - check if any side effects
TODO: Test torrent quick replacement on blockchain data corruption

rootzoll, 2019-04-25:
This feature is a bit half backed and just an experimental part of the v1.2 release. I was not able push Torrent Download/Seeding enough into the background so that CPU and USB-HDD-Bandwith keep making problems.

fluidvoice, 2019-04-26:
My first look, it seems many are using the **Deluge** torrent client which is supposedly light weight run as a daemon with remote UI/access.  Needs more research.

rootzoll, 2019-04-27:
@fluidvoice I reasearch into alternatives to rtorrent like Deluge may make sense for a future release.

For v1.2 I think I leave it as it is to experiment with. So I move the ticket to finish to a future realease.

rootzoll, 2019-09-09:
Now with RP4 the torrents get less and less important. Seeding from the raspiblitz in the past did not worked well and so I think OK to burry this idea.

closed 2019-09-09 by rootzoll