raspiblitz/raspiblitz #28: Stuck post Blockchain Download

evilgrinz, 2018-07-31:
UNKOWN STATE
It seems that something went wrong during sync/download/copy of the blockchain.
Maybe try --> ./60finishHDD.sh

Do i need to just re-dl it? Is it just timing out trying to sync? I enter this command, and it just pops again after 5 minutes.

rootzoll, 2018-08-01:
Make sure ta have the latest SD-Card image - checl README.md
Quick solution would be to delete your blockchain directory: sudo rm -r /mnt/hdd/bitcoin
The make a fresh reboot.

evilgrinz, 2018-08-01:
ill give it a go, and refresh the image thanks

charlesrocket, 2018-08-04:
looks like issue is fixed with the fresh image

rootzoll, 2018-08-04:
thx, so I will close this issue fo now

closed 2018-08-04 by rootzoll

rootzoll, 2018-08-06:
How SSH Works:
https://www.youtube.com/watch?v=zlv9dI-9g1U

For intro into Raspi/Linux Terminal ... Hack5 has a quite entertaining show lineup:
https://www.youtube.com/watch?v=b5NmtmNwMgU&list=PLW5y1tjAOzI2ZYTlMdGzCV8AJuoqW5lKB