raspiblitz/raspiblitz #35: failure while unlocking lnd wallet

philippeowagner, 2018-08-08:
`[lncli] rpc error: code = Internal desc = connection error: desc = "transport: authentication handshake failed: x509: certificate signed by unknown authority"`

I tried https://github.com/mably/lncli-web/issues/151 without success.

Any suggestion of fixing this? 
Is it possible to re-install lnd?

philippeowagner, 2018-08-08:
```
 admin@raspberrypi:~ $ lncli create
Input wallet password:
Confirm wallet password:

Do you have an existing cipher seed mnemonic you want to use? (Enter y/n): n

Your cipher seed can optionally be encrypted.
Input your passphrase you wish to encrypt it (or press enter to proceed without a cipher seed passphrase):

Generating fresh cipher seed...

[lncli] unable to generate seed: rpc error: code = Internal desc = connection error: desc = "transport: authentication handshake failed: x509: certificate signed by unknown authority"
```

rootzoll, 2018-08-08:
can you try 2 things:

1) `su bitcoin` and see if you stil get the error
2) copy macaroon and tlscert fresh from /home/bitcoin/.lnd to /home/admin/.lnd

let me know what the results are

philippeowagner, 2018-08-08:
The first option worked. I could recreate using `lncli create`. 

This leads to the following questions:

1. Where is the LND CIPHER SEED stored if this process is done automagically?
2. After rebooting the node,  the admin's wallet is tried to be unlocked.

rootzoll, 2018-08-08:
1) There is no automagically creation of the LND wallet. During raspiblitz setup the lncli create is just triggered for the user .. the user then has to go manually thru the wallet creation dialog and is asked by the raspiblitz to write down the LND CIPHER SEEDS before continue.

2) On every reboot, the LND Wallet needs to get unlocked. The RaspiBlitz is just offering to start the unlocking and calls lncli unlock for you if you choose the main menu option. But the password entering has to be done manually within the LND unlock dialog.

Hope I understood your questions correct.

philippeowagner, 2018-08-08:
Thanks for the hints. It works now as expected.

closed 2018-08-08 by philippeowagner

wuestenigel, 2019-01-09:
I'm getting this error while connecting via Joule Extension:

> authentication handshake failed

Anyone have an idea how I can solve this? Don't want to recreate seeds and so on as my node is running very well.