raspiblitz/raspiblitz #43: Add a Admin WebUI

rootzoll, 2018-08-15:
To have a WebUI the user can call to control the RaspiBlitz from Desktop or Mobile with more detailed options would be great ... for example:
https://github.com/ShahanaFarooqui/RTLFullApplication

rootzoll, 2018-12-24:
When doing a Admin WebUI .. also integrate Zap URL Connet. See ticket #174

felivix, 2019-03-21:
@rootzoll , Raspiblitz works well and I use RTL to interact with it. Thank you!

I like to customize a few things and from the console installed mc and some other useful software. I also tried to initialize an X session with sudo startx but the process got to a halt. 
Q: is it possible to run an X session on the raspiblitz?

fluidvoice, 2019-03-21:
> @rootzoll , Raspiblitz works well and I use RTL to interact with it. Thank you!
>
> I like to customize a few things and from the console installed mc and some other useful software. I also tried to [...]

Top response in my first search:
https://raspberrypi.stackexchange.com/questions/43847/startx-command-not-found

openoms, 2019-08-02:
Flask based web interface alternative: https://github.com/M-D-Br/FlaskBitcoinDashboard

rootzoll, 2021-03-17:
Also take look at this WebUI by @Kixunil : https://github.com/rootzoll/raspiblitz/issues/2056#issuecomment-800908527

Kixunil, 2021-03-18:
FYI, I also intend to create "pico LND manager" - it will show the seed and LND Connect.

openoms, 2021-03-20:
I have good experience using the Debian flavor called Freedombox: freedombox.org.
Demo: https://freedombox.org/demo/
The web UI is written in python which makes it accessible (although I am not familiar with the code yet): https://salsa.debian.org/freedombox-team/freedombox/-/tree/master/plinth
And of course being Debian all of it is Free Open Source under the: GNU Affero General Public License, Version 3
https://salsa.debian.org/freedombox-team/freedombox/-/blob/master/COPYING.md

Emzy, 2021-03-27:
Must work in Tor-Browser.
Also no nodejs please.

Kixunil, 2021-03-27:
@Emzy `selfhost-dashboard` works in Tor Browser just fine (@MaxHillebrand can confirm) and has no nodejs stuff. It will use some reasonable libraries in the future (Vue, I think) but I don't intend to go crazy with it. I'm **not** implementing frontend myself, other people do, but I regulate them. :)

Additional advantages: it **already** supports SSO for RTL, ThunderHub and btc-rpc-explorer. It will have SSO support for BTCPayServer once it's implemented on their side. Generally, if you can get sign-in link to an app using a shell script/program, then SSO is supported. OAuth seemed too bulky to implement and designed for a different use case but since many projects use it maybe it will be implemented eventually.

The UI looks a bit like freedombox, but I consider it nicer and more modern, which is needed to avoid deterring people who will use vulnerable projects instead... Freedombox doesn't have existing Bitcoin integration, so that'd have to be coded.

rootzoll, 2021-03-28:
Also to take a look at this prototype Dashboard by @fusion44 https://github.com/fusion44/blitz_dashboard

rootzoll, 2021-03-28:
For design resources and community to connect to ... also reaching out to https://twitter.com/bitcoin_design is always welcome :)

cstenglein, 2021-03-30:
For the Web UI I started working on it [here](https://github.com/cstenglein/raspiblitz-web) with React & Tailwind CSS.

Since I don't know about RPC / Protocol Buffers & my Backend skills are currently limited (only some Java & a bit NodeJS), it could take some weeks for the backend (most likely written in go, since that would be my next goal to learn). If go is not fine, I could also try making myself familiar with python & FastAPI, fine with me either way.

At first it would just be the pages shown in the screens (maybe I'll take services out at first for the MVP). 

Later it could be expanded with a setup wizard (so the RapiBlitz can be set up with it & installing the additional services but I don't want to promise too much and first deliver something.

cstenglein, 2021-03-31:
Good point!

Are you talking about the frontend or backend?
For backend I am happy to hook up to anything, actually.
I looked at the umbrel and RTL code which only use Node.js which isn't is not liked in the core code.

For the frontend:
- the flask dashboard didn't have a license. 
- the `blitz_dashboard` is meant for the display if I see that correctly?
- ~~The `selfhost-dashboard` of your is a backend only currently, isn't it?.~~ nvm, now I got it, sry.  :man_facepalming: 

Still not sure about your [comment](https://github.com/rootzoll/raspiblitz/issues/43#issuecomment-808804297) with implementation though?

For me it's a nice learning experience to do something like this so I wouldn't mind some extra effort.

fusion44, 2021-03-31:
The dashboard in its current state can only run on the pi itself as a
native app. Once we have an abstraction in form of a backend it could be
compiled for the web and served as a webpage. But my personal use case is
foremost a dashboard in a bigger screen.

I'm currently on the road. I'll write more about my plans later this
evening 😃

2021, 07:19:

Kixunil, 2021-03-31:
@cstenglein yeah, I was talking about both. I wrote the backend which is designed for security-critical application (but I still need to do more work on it to be confident in claiming it's secure).

Perhaps it'll be useful to share a screenshot:

[image]

The screenshot shows all the available applications, including one WIP secret app. (I prefer to test it properly privately before a bunch of newbies generates support tickets.) If you like the app, please do **not** promote it as I'd like to have  bit more time for development before too many people use it.

The list of applications is not fixed, it's dynamically generated. You can easily add another app just by adding a few simple files in the filesystem. This is being performed automatically from installation scripts of other packages in my repository.

> Still not sure about your comment with implementation though?

I'm not sure I understand. I guess you're unsure how development works now.
The dashboard was quickly developed by a friend, who doesn't have that much time lately so another experienced friend will help with it. I could get you in touch with them if you're interested but I'm a bit afraid of having too many devs on a single project would *decrease* efficiency. May still be worth trying to figure out if it can be done efficiently or not.

rootzoll, 2021-03-31:
To sync up with @kaloudis once we research on Backend/Frontend ... because RoninDojo is also planing on an abstract layer for lightning implementation .. see https://twitter.com/BTCxZelko/status/1377335920026787844?s=20 we could reuse here.

Kixunil, 2021-04-01:
@rootzoll that seems to be not entirely related to admin UI, which would presumably manage more things than just LN? Am I missing something.

openoms, 2021-04-01:
Yes, RTL and the Zeus app are doing the lightning implementation agnostic LN management pretty well already.
So unless one of those are planned to be accomodated to the blitz more we are just needing an interface to launch/pair them in the webUI.

Kixunil, 2021-04-01:
RTL is already supported by `selfhost-dashboard` and Zeus is just lndconnect, right? I'm working on proper lndconnect UI right now. :)

Frankly, last time I checked Zeus, it didn't seem "pretty well" but maybe it has changed. Zap is LND-only but I had much better experience.

kaloudis, 2021-04-01:
@Kixunil you may have had some problems connecting since Zeus uses REST whereas Zap uses GRPC

I'm willing to help with this effort where I can. Are there plans on using React or something else?

Kixunil, 2021-04-02:
Hmm, that could be it.

Do you mean `selfhost-dashboard`? That one is currently quickly hacked together but will use Vue as I've been told is lighter than React and thus probably better for security

kaloudis, 2021-04-05:
Vue is a good choice too

rootzoll, 2021-04-21:
Just as an idea reminder and for discussion: I think its a good UX to signal that your basic node setup is done (confetti moment) early and not delay that until the blockchain is 100% synced.

Sure some apps and functions (like opening channels) will need a fully synced blockchain ... but basically thats just waiting time and we should test what a dashboard can already offer before that. How can we make good use of that waiting time in the end?

Here are some ideas to check:

- Can we do already a on-chain funding transaction to LND? That would be great - because we can save on fees then, we are not in a hurry because syncing of blockchain will take +2days anyway.
- Also settingup/learning on backups, watchtowers, should also already be possible
- We can offer educational content ... videos explaining Basics on the Lightning Network, etc

All this does not need to be in the v1.8 from the beginning - but it might make sense to give basic access to the web dashboard eraly on and while some parts wait for the blockchain to sync, the node already feels operational & usefull.

Kixunil, 2021-04-21:
Good points, my Dashboard doesn't force people to wait, it's not even aware of Bitcoin. Adding some comforting message sounds interesting.

> Can we do already a on-chain funding transaction to LND?

You can send it safely from your own wallet, that makes sense. OTOH I was exploring (ab)using PayJoin to fund channels directly from your own wallet.

I plan to eventually package a handful of non-Bitcoin privacy/freedom related apps, so those could be useful meantime. You can also configure a few things while waiting - e.g. setup XPUB/hot wallet in BTCPayServer, setup stores, apps...

rootzoll, 2021-04-28:
Hi everyone - now that v1.7 is out I am sorting the options for the RaspiBlitz v1.8 WebUI. First I try to get an overview.

Just to recap a bit what I see as important for the WebUI (list not complete/final):

- we have Setup-Process & Dashboard/Settings (& maybe a kind of AppStore) to cover with a WebUI - if we can utilize also something for the LCD/TouchUI from thats fine, but that can work basically indepently from WebUI and is a different task for v1.8
- it does/should not need to care about lightning management - because other apps like RTL or Thunderhub, etc can do that part
- the backend will try to introduce some abstraction/modular structure so that we can have more lightning implementation options (e.g. c-lightning, etc) in the future - the frontend needs to reflect this
- it should use i18n features that will work with services like Transifex.com so that it can be translated into different languages later
- framework needs to work in Tor browser and should not depend on outside-loading dependencies
- frameworks with reduced & wellchecked dependencies are preferred to have a minimal attack surface
- also I need to be able to work in that framework to be able to fix, maintain & also create/integrate features .. I am open to new stuff but you know the problem with old dogs and new tricks ;)
- the more someone can commit personal time ressources to build/implement on the frontend side the more I can follow personal framework choices - if needed we maybe have even a small budget to compensate for open-source development on the frontend side

@Kixunil I was shortly looking into https://github.com/debian-cryptoanarchy/selfhost-dashboard ... I can see its Rust based, but a bit hard to see what other frameworks are used for frontend. Have to say I am a total Rust noob. Can you give more intro/details?

@cstenglein https://github.com/cstenglein/raspiblitz-web looks like a nice start for a React based frontend. Do you have bigger time ressources within the next two month just in case?

@fusion44 dart & flutter as shown https://github.com/rootzoll/raspiblitz/issues/881 seems a bit like a bet on googles open-source efforts but update us in your latest progress

@hashes4merkle has a PR to add the Homer-Dashboard to the RaspiBlitz - that also could be maybe an option to look into https://github.com/rootzoll/raspiblitz/issues/2162 if it can be extended with web components

@arcbtc building a WebUI/backend with a similar stack like LNbits could also connect well from a dev community side

Kixunil, 2021-04-28:
Frontend is currently vanilla JS. We plan to migrate to Vue eventually. I believe it provides translation integration as well.

* Setup process - what kind of setup is required in Raspiblitz? My dashboard only needs setup that can be completely automated (it's automated in deb packages now), anything else can be an app on top of it
* "app store" - is planned as a separate app that will "just" forward to `apt`.
* LN, even Bitcoin is completely independent of my dashboard.
* Modular architecture is already supported
* Dashboard currently works in Tor browser and will continue to work. We don't have no-JS requirement because  preventing people from attacking themselves is pointless. Outside dependencies are indeed forbidden.
* To my knowledge Vue is not bloated.
* I intend the dashboard to be used in professional product and thus you can expect good amount of development time. I hope the differences between Deb repo and Raspiblitz will not be too great to be a problem.

fusion44, 2021-04-28:
Warning: Long wall of text ahead with my brain-dump on this topic.

Since the last Hackday I've been learning how the setup process of the Raspiblitz works under the hood to better asses the requirements we have to have on a server. 

I see two viable paths we can go with the backend side of things. Basically I'm comparing adding a server to the current bash script based setup with a selfhost-dashboard based approach. They are quite different and both have advantages and trade-offs.

### A: Keep the current bash script based setup flow.
* Works with the existing Raspiblitz code base
* UI code (whiptail) and system management code is currently very intertwined. This makes a refactor necessary to properly separate the system and UI layers. 
* A server component is necessary which in turn would call the bash scripts which in turn then executes the system level tasks. These system level scripts should not show UI dialogs directly.
* There would be two ways of notifying the server of changes on the system:
  * The scripts call a REST endpoint with a message what has changed. 
  **Advantage**: The server always knows what is going on on the system.<br>
  **Disadvantage**: UI layer and system layer are again somewhat tightly coupled.
  * The server only receives data when it actively calls a script. The script should then return data in JSON format. Server and system may go out of sync.
  **Advantage**: UI layer can be neatly separated from the system level scripts<br>
  **Disadvantage**: Server mey go out of sync when user changes something from the SSH menu.

Based on a server I've created for another personal project I've implemented a very simple server in Golang + GraphQL. Please note that this is very early and only for toying around with different ideas:
https://github.com/fusion44/blitz_backend

The current simple graphql schema file:
https://github.com/fusion44/blitz_backend/blob/main/graph/schema.graphql

Besides the user management it just has little but important functions currently:
* A call to notify the server of changed device information
* A query to get current device information
* A subscription to push changed device information to subscribers

The device information basically consists of all information in the raspiblitz.info file. As I understand it this file is used to "communicate" the current device state between different scripts.

I've also created a small Desktop app to test the server but it currently looks ugly and is not compatible with the current GQL schema.

### B: Build Raspiblitz with selfhost-dashboard
* as far as I can see, selfhost-dashboard works a bit like Umbrel but uses deb packages instead of Docker and Rust instead of NodeJS. The packages are installed, run and then wired together by the Rust code. @kixunil please correct me if I got this wrong :)
* User management is already build in
* I don't see how the current bash script based approach can be easily merged with selfhost-dashboard as the Rust code is the replacement for the bash scripts
* This approach seems a bit more modular as with the script based approach.

To be fair, I'm way less familiar with Rust and selfhost-dashboard, but it seems that the choice here is more an either-or rather than a merger of two projects. I don't yet have a clear preference since both have their distinct (dis)advantages.

My only personal requirement is that I can use the Flutter framework as I'd like to take a deep dive into it while building this project.

As an aside:
To quickly boot up and tear down new Raspiblitz instances I've created a simple [Docker image](https://github.com/fusion44/raspiblitz/tree/docker_dev/alternative.platforms/amd64/docker). This is not **meant to be used in production**. Basically it is a stripped down Raspiblitz setup procedure with a few scripts like formatting the hard drive disabled. It can be used to quickly test change in scripts. I understand there is the Vagrant platform but I had trouble getting it work.

fusion44, 2021-04-28:
To your points @rootzoll
> * we have Setup-Process & Dashboard/Settings (& maybe a kind of AppStore) to cover with a WebUI - if we can utilize also something for the LCD/TouchUI from thats fine, but that can work basically indepently from WebUI and is a different task for v1.8

Flutter can be compiled as a native app so it can be run directly on the Raspiblitz as an optional app. I haven't yet gotten Flutter to work with the default Screen for Raspiblitz but I think this is because during the build of the SD card image the egl drivers are disabled.

> * it does/should not need to care about lightning channel management - because other apps like RTL or Thunderhub, etc can do that part

I have this functionality in SendMany. I can port it over or not. My personal long term goal is to make a bitcoin education station with different resources (Maybe this should even be a whole different app).

> * the backend will try to introduce some abstraction/modular structure so that we can have more lightning implementation options (e.g. c-lightning, etc) in the future - the frontend needs to reflect this

IMHO this should be a separate project. Maybe even investigate a similar project to avoid redundancy: https://gitlab.com/inbitcoin/lighter

> * it should use i18n features that will work with services like Transifex.com so that it can be translated into different languages later

Yes. I have successfully tested the translation library I use with SendMany with Weblate.

> * framework needs to work in Tor browser and should not depend on outside-loading dependencies

I'd have to try that but I don't see why Flutter should have problems with that.

> * frameworks with reduced & wellchecked dependencies are preferred to have a minimal attack surface

A certain level of scrutiny when introducing new dependencies is always advisable. Although Flutter has many features build in, you probably can't build an app completely without external packages.

> * also I need to be able to work in that framework to be able to fix, maintain & also create/integrate features .. I am open to new stuff but you know the problem with old dogs and new tricks ;)

> * the more someone can commit personal time ressources to build/implement on the frontend side the more I can follow personal framework choices - if needed we maybe have even a small budget to compensate for open-source development on the frontend side

Currently I should be able to spend around 4-5 hours on the project per week. When the weather sucks maybe even more.

cstenglein, 2021-04-28:
On my part, I haven't progressed very much from the last month ago, still learning (underestimated the effort in learning too much stuff at once). I cut out many things to have a working MVP soon.

Regarding your points I got a couple of questions:

> we have Setup-Process & Dashboard/Settings (& maybe a kind of AppStore) to cover with a WebUI

Thanks for the clarification! Makes stuff easier to focus on for the moment. 

> we have Setup-Process

I'm currently unsure how the setup process should work in conjuction with the bash scripts (setting the passwords etc). 
Maybe it would be possible to trigger the installation scripts from the backend (e.g. with args to skip the prompts)?

I'll currently focus on a dashboard which shows the status after the setup process is done atm since I haven't found an answer on that yet.
(P.S. While writing the post, @fusion44 made some elaborate thoughts on that, will read later)

> (& maybe a kind of AppStore) 

Currently the installation is triggered from the scripts with the "bonus" prefix AFAIK. So would it be enough for the app store to show certain preconfigured apps (e.g. a JSON with the Name, description, the script & optional address) and trigger the installation script from the UI? 
The optional address would be to have a link of an already installed app (e.g. localhost:12345) to ping it for a online/installed status and create a link when it is already installed.
(P.S.: looks like umbrel does something similar https://github.com/getumbrel/umbrel/blob/master/apps/registry.json )

> it does/should not need to care about lightning channel management - because other apps like RTL or Thunderhub, etc can do that part

I guess funding the LN Wallet and showing number of channels / balance should still be in the UI?

> the backend will try to introduce some abstraction/modular structure so that we can have more lightning implementation options (e.g. c-lightning, etc) in the future - the frontend needs to reflect this

The raspiblitz uses `lnd` at first, right? So that would be the default and afterwards it should be configurable in the settings?

> framework needs to work in Tor browser and should not depend on outside-loading dependencies

What are outside-loading dependencies? Web Components?

> Do you have bigger time ressources within the next two month just in case?

What are "bigger time resources"? :)
I realized I should first learn more in depth about React & Go before jumping in it as I was doing at the beginning where I was jumping around too much. It seems like there is not much code progress but my current focus is only the dashboard, so no worries there. :)
 
For inclusion into RaspiBlitz I'll gladly let @fusion44 & @Kixunil go ahead since I don't want to provoke some competitive situation and just want to focus on some nice dashboard I would gladly use as a non-technical user.

rootzoll, 2021-04-28:
I will read in detail tomorrow but just some quick clarification on the Setup process and how it can work with a WebUI.

At the moment the setup needs you to SSH in and run the install scripts actively in your SSH session. This is to be changed with v1.7.1 in preperation to the v1.8 release - see #1126 - so basically the raspiblitz will start up, run a some bootstrap shell scripts to do some initial things/reboots like extending filesystem of sd card, etc and then analyse the system. Gathering all the information needed for a setup process. In v1.7.1 you will still login per SSH go thru a lot of dialogs at once (not as interactions during the setup process) and that will create a "raspiblitz.conf" with some additional setup info, store that and kick-off the re-factored setup shellscripts that go thru the complete setup non-interactivly (like a recover/update process right now) .. and present things like your seed words to finish up setup before it goes into blockchain sync.

With v1.8 introducing a WebUI for setup ... there will be a setup WebUI after the system did its basic information gathering and as an alternative to the SSH dialogs (that still will be there for people that prefer) the WebUI will have a form/dialog creating the same "raspiblitz.conf" with some additional setup info and kicking off the non-interactive setup process while giving status updates (like you see on recovering/update on the LCD) until its done. 

Sorry for not layout that re-factorin idea more clearly before. Its still a concept, but will be done with v1.7.1 in preperation for v1.8.

rootzoll, 2021-04-28:
I think it would help to have Jitsi session again to exchange on the topic this week - i created a doodle to find a time that fits most - @cstenglein @fusion44 @Kixunil please let me know what date/time would be possible for you. thx: https://doodle.com/poll/dp2qtu25y57zrdhw?utm_source=poll&utm_medium=link

Jitsi room would be: https://meet.fulmo.org/raspiblitz-web-ui

Kixunil, 2021-04-29:
@fusion44 I fail to see any note worthy resemblance to Umbrel. `selfhost-dashboard` is completely Bitcoin-agnostic, their UI is Bitcoin-first. Yes, both use some webserver and HTML frontend but that could be compared to any other website as well.

Since selfhost-dashboard can be installed completely interactively (using deb package is the easiest) the simplest way to use it is to just install it right away and then add a special setup app for Raspiblitz. I plan to implement some settings that will be tied to `dpkg`/`apt` as a separate app. Raspiblitz can either move to deb repo or use its own app.

There's currently minimal user management - just admin. The database is there but no simple way of adding another user. It can be implemented easily though.

> the Rust code is the replacement for the bash scripts

I don't think this is accurate. Rust is just a server. It's more like `apt`/`dpkg` are replacement for bash scripts in my repository, but it's orthogonal to dashboard.

> I can use the Flutter framework

For Raspiblitz-specific app I don't care, for the dashboard itself and app store I will only accept what my very reliable, skilled developer chooses (he seems to prefer Vue). I have some important requirements that make him the best person to implement it.

> either-or rather than a merger of two projects

I actually don't see a significant issue preventing using both if Raspiblitz-specific stuff can be made as a separate app. Dashboard only handles opening of other apps. I can make slight changes like a hook for setup app that will open after creating the admin account.

rootzoll, 2021-04-29:
@fusion44 @cstenglein @Kixunil we have a date/time thats possible for evrybody:
Sunday 18:00 --> https://meet.fulmo.org/raspiblitz-web-ui

rootzoll, 2021-04-29:
This is a first draft of how the re-factored setup process starting in v1.7.1 with just SSH dialogs and then in v1.8 with additional WebUI could work:

[image]

Source is here and open for comments:
https://docs.google.com/drawings/d/1iYYXrZTFqH5xMVxSXP7BSmf38vIj6GhXtJoVGFCHrQE/edit?usp=sharing

cstenglein, 2021-05-03:
As suggested by @fusion44 I started moving to websockets and creating a mock backend with NodeJS which just sends some static data: https://github.com/cstenglein/raspiblitz-web/tree/master/backend-mock

Getting and changing the state in real-time is awesome and works, e.g. when getting the syncstatus [here](https://github.com/cstenglein/raspiblitz-web/blob/master/backend-mock/bitcoin.js#L13-L25).

Currently, I got 2 issues with this approach:

1. I have one socket,  but through this I get many different types of data (syncstatus, transactions, receive payment & send payment currently), so I have to differentiate between these requests with an ID as fusion44 suggested yesterday. Still, I'm not sure if this is the right approach going forward (see [here](https://github.com/cstenglein/raspiblitz-web/blob/master/src/components/Home/Home.tsx#L21-L55) on what I mean).

2. As discussed yesterday, there will be a web api in python soon but it's not yet defined. Should I document somewhere what data I'll need for the different calls? Would be nice to have some certainty beforehand.

FYI @rootzoll

fusion44, 2021-05-03:
Going 100% websocket is probably not the best idea, as it makes things impossible for shell scripts to call the API. I might have made it sound like it's my preferred way to go yesterday, sorry :)

Actually having event IDs as a string might be preferable over numeric event IDs:
https://thoughtbot.com/blog/json-event-based-convention-websockets

Lets imagine an event where the setup must stop and ask for user input. Lets take DNS Enter/Select as an example. This is how it currently looks in the SSH setup menu:
[image]

We may like to give the user multiple options:
* ignore
* default (1.1.1.1)
* custom

From the backend:
```JSON
{
  "setup_dns_invalid": {
    "message": "Failed to resolve DNS requests",
    "default_options": [
      {
        "ip": "1.1.1.1",
        "name": "Cloudflare",
        "link": "https://cloudflare-dns.com"
      },
      {
        "ip": "9.9.9.9",
        "name": "Quad9",
        "link": "https://www.quad9.net"
      }
    ]
}
```

At this stage the server waits for an answer.

Here's a possible answer from a clients:
```JSON
{
  "setup_dns": {
    "action": "change, default or ignore",
    "ip": "1.1.1.1 <- must not be null when change is set"
  }
}
```
The advantage of a string ID is that it's human readable. This helps code clarity. 
I think that the clients answer should probably be a rest call instead for the reason stated above.

cstenglein, 2021-05-08:
Wanted to give you an update:

Last week I focused on the Dashboard page and made some adjustments there. I am currently using a mock for the backend so the data is more or less static (some setTimeout to test some stuff). 

The screenshots can be found [here](https://github.com/cstenglein/raspiblitz-web/tree/master/images/2021-05-08).

Current next steps:

- Send Dialog: Validation for entering a correct address, not spending more than balance etc + better design
- Implement Receive & Send for Lightning Box as well
- Add Settings Page (Reboot / Shutdown / Change Password currently)
- Add App Page (need to think about)
- Add Authentication & Login
- Think About how to implement the Setup Process
Later: 
- Add Translations (Still have to figure out Transifex, but made an account there and added the project)
- Maybe add 4th Widget (because in Desktop / small Desktop there is much space left)

rootzoll, 2021-05-23:
Notes from Call 23.05.2021:

- backend will just have static roles: public, webadmin, touchscreen
- Authentifizierung: Basic Auth für den Anfang
- WebUI uses Password A (username is not needed, may check with password managers & Tor browser)

rootzoll, 2022-05-09:
This will now finally come with the v1.8.0 release :)
See progress on on the WebUI repo: https://github.com/cstenglein/raspiblitz-web

rootzoll, 2022-06-05:
To be tested during RC2

rootzoll, 2022-07-27:
OK final tests were good and WebUI is in 1.0.0 state for v1.8.0 release. Closing issue.

closed 2022-07-27 by rootzoll