raspiblitz/raspiblitz #45: configuring firewall

renepickhardt, 2018-08-15:
I was looking to see if raspiblitz has a firewall activated but realized you just install it but don't configure it.

https://github.com/rootzoll/raspiblitz/blob/ee67947a631926779783d4cc22127108b4d8e284/build.sdcard/raspbianStretchDesktop.sh#L92

for basic usage (bitcoin + lightning) the following patch should fix this:

```
ufw allow ssh
ufw allow 8333
ufw allow 9735
ufw default deny incoming
ufw default allow outgoing
ufw enable
```

rootzoll, 2018-08-16:
@renepickhardt totally correct. Now that people start to use the RaspiBlitz at home (not just for hacking on a event) we need to add firewall config. 

@vnnkl thx for the PR :) ... hope to integrate it till the end of this week.

renepickhardt, 2018-08-18:
hey @vnnkl that is my PR!  srsly: I already provided the patch to the correct line
@rootzoll feel free to merge anyway (:

vnnkl, 2018-08-19:
@renepickhardt 
I know, i referenced it in the pr. did not realize you wanted to open a pr yourself beyond this issue with pointers on where you'd put the changes, feel free to go ahead and open one to be merged

renepickhardt, 2018-08-22:
@vnnkl no problem I already said above that rootzoll can merge your request. so everything is cool. Hope to meet you at the lightning hackday!

closed 2018-08-22 by rootzoll

rootzoll, 2018-08-24:
OK. Thanks all. I added firewall to the end of the configuration process. Do anybody think, having the firewall up after the whole sync/scan process is a problem?

renepickhardt, 2018-09-04:
@vnnkl as discussed on the lightning hackday:
```
curl -i -H "Accept: application/json" -d '{"amount":ENTER_AN_INTEGER_AMOUNT_OF_SATOSHIS_INSTEAD_OF_THIS}' http://ln.rene-pickhardt.de/invoice
``` 
@rootzoll : why not doing it first?

vnnkl, 2018-09-04:
@renepickhardt succeeded on 2nd attempt √

PR #47 (2018-08-22): configure ufw
added configuration as suggested by #45 
+ testnet

closes #45