raspiblitz/raspiblitz #46: other lightning implementations

renepickhardt, 2018-08-15:
As far as I see the raspiblitz currently only runs with lnd. is there any reason (other then manpower) to not include c-lightning and have the user prompted which implementation they want to run? If there is no reason I might take the lead to also include c-lightning.

rootzoll, 2018-08-16:
Having both to choose/switch between LND and c-lightning would be great ... related to manpower we just ran with LND in the beginning. 

@renepickhardt would be great if you can take the lead on c-lightning integration. We maybe have to see if it makes sense to restructure the shell-scripts a bit for that, so that we dont get too much into a if/then hell. Maybe first step could be to put all LND related shell script fragments into a kind of Lightning-Service Shell-Library: https://bash.cyberciti.biz/guide/Shell_functions_library .. and then in a second step to make this Library dynamically work with LND or c-Lightning were possible. Do you think that makes sense?

renepickhardt, 2018-08-18:
I think it does. I realized most of the stuff is shell script which is not necessarily my favorite language. so if you are strong on shell script I would rather have you to restructure the scripts and basically create an abstraction layer for interaction with the lightning node. once that is done I could easily do all the c-lightning related stuff. or we just do it together before , while or after the lightning hackday in berlin (:

rootzoll, 2018-08-19:
I think thats a good task for the hackday :) ... lets go thru the scripts together and see it the steps that are in there for LND make also sense for c-lightning or if it may need even more rearrangement, than just an abstraction layer.

renepickhardt, 2018-08-22:
bsss I am thinking all the time that it might by cool to buy a raspiblitz. now I get even more tempted...

rootzoll, 2018-09-13:
Some guide to look into: https://medium.com/@meeDamian/c-lightning-node-on-rbp3-b950660fb835

rootzoll, 2018-12-24:
https://twitter.com/KHS9NE/status/1042837790822727680

nyxnor, 2021-04-30:
@renepickhardt 
> If there is no reason I might take the lead to also include c-lightning.

Hi Mr Pickhardt.

nyxnor, 2021-05-11:
@rootzoll Please close this, Openoms continued here https://github.com/rootzoll/raspiblitz/issues/2295#issue-886867501

openoms, 2021-11-08:
C-lightning is implemented. Closing this issue for inactivity.

nyxnor, 2021-11-08:
I remember fiatjaf posting about `wen eclair` on the last blitz release, but if noone will do it, then ok.

openoms, 2021-11-08:
Sure and there is the Rust LDK too.
If someone starts to work on these will worth it's own issue. Even it's own project tbh (fork of this one or not).

closed 2021-11-11 by rootzoll