raspiblitz/raspiblitz #51: BUILD FAILED --> Bitcoin download checksum not OK

Smiggel, 2018-08-20:
Trying to setup a node the manual way. Installing using the script works fine until it should check the code of Bitcoin and it stops with the error "BUILD FAILED --> Bitcoin download checksum not OK".

Anyone else having this issue?

rootzoll, 2018-08-20:
Thats line 131 in `rasbianStretchDesktop.sh` .. can you run and post what the output is?
`sha256sum --check SHA256SUMS.asc --ignore-missing`

Smiggel, 2018-08-21:
Not sure what you need. This is what i see on my terminal:

> sudo: kan computernaam raspberrypi niet herleiden
mkdir: kan map ‘/home/admin/download’ niet aanmaken: Bestand bestaat al
sudo: kan computernaam raspberrypi niet herleiden
--2018-08-21 09:01:56--  https://bitcoin.org/bin/bitcoin-core-0.16.2/bitcoin-0.16.2-arm-linux-gnueabihf.tar.gz
Herleiden van bitcoin.org (bitcoin.org)... 138.68.248.245
Verbinding maken met bitcoin.org (bitcoin.org)|138.68.248.245|:443... verbonden.
HTTP-verzoek is verzonden; wachten op antwoord... 200 OK
Lengte: 9829750 (9,4M) [application/octet-stream]
Wordt opgeslagen als: ‘bitcoin-0.16.2-arm-linux-gnueabihf.tar.gz.3’

bitcoin-0.16.2-arm-linux-gnueabihf.tar. 100%[==============================================================================>]   9,37M   202KB/s    in 48s     

2018-08-21 09:02:45 (202 KB/s) - '‘bitcoin-0.16.2-arm-linux-gnueabihf.tar.gz.3’' opgeslagen [9829750/9829750]

sudo: kan computernaam raspberrypi niet herleiden
--2018-08-21 09:02:45--  https://bitcoin.org/bin/bitcoin-core-0.16.2/SHA256SUMS.asc
Herleiden van bitcoin.org (bitcoin.org)... 138.68.248.245
Verbinding maken met bitcoin.org (bitcoin.org)|138.68.248.245|:443... verbonden.
HTTP-verzoek is verzonden; wachten op antwoord... 200 OK
Lengte: 1957 (1,9K) [application/octet-stream]
Wordt opgeslagen als: ‘SHA256SUMS.asc.3’

SHA256SUMS.asc.3                        100%[==============================================================================>]   1,91K  --.-KB/s    in 0s      

2018-08-21 09:02:46 (4,50 MB/s) - '‘SHA256SUMS.asc.3’' opgeslagen [1957/1957]

sudo: kan computernaam raspberrypi niet herleiden
--2018-08-21 09:02:46--  https://bitcoin.org/laanwj-releases.asc
Herleiden van bitcoin.org (bitcoin.org)... 138.68.248.245
Verbinding maken met bitcoin.org (bitcoin.org)|138.68.248.245|:443... verbonden.
HTTP-verzoek is verzonden; wachten op antwoord... 200 OK
Lengte: 17940 (18K) [application/octet-stream]
Wordt opgeslagen als: ‘laanwj-releases.asc.3’

laanwj-releases.asc.3                   100%[==============================================================================>]  17,52K  --.-KB/s    in 0,03s   

2018-08-21 09:02:47 (680 KB/s) - '‘laanwj-releases.asc.3’' opgeslagen [17940/17940]

!!! BUILD FAILED --> Bitcoin download checksum not OK

Smiggel, 2018-08-21:
Ok, figured it out. I was in the wrong folder. :-)

I get this:
bitcoin-0.16.2-arm-linux-gnueabihf.tar.gz: good
sha256sum: WARNING: 20 lines are incorrectly formatted

Smiggel, 2018-08-21:
Issue is fixed. Tried again. Seems like it went wrong because I stripped Raspbian from packages I don't use. Doing it without the stripping it works fine.

closed 2018-08-21 by Smiggel