raspiblitz/raspiblitz #54: secure and test pi user

rootzoll, 2018-08-24:
At the moment the raspiblitz makes an auto login on powerup with user pi to play info to the display. At the moment this user has too much rights. If some gets physical access to the device, you just need to connect a keyboard and you own it.

TODO: Limit rights of unix user pi, use readonly.macaroon with lncli and test what this user could still do.

vv01f, 2018-08-26:
one uncomplicated thing to limit the user is a restricted shell, e.g.

`sudo chsh pi -s /bin/rbash`

another thing is to prevent him from exiting the script, this can be done with [exec](http://wiki.bash-hackers.org/commands/builtin/exec), [trap](http://wiki.bash-hackers.org/commands/builtin/trap) is an alternative for fallback … :

```
# automatic start the LCD info loop
SCRIPT=/home/admin/00infoLCD.sh
# trap may not be necessary when exec is used
trap $SCRIPT EXIT
# replace shell with script => logout when exiting script
exec $SCRIPT
```

[further limits](https://www.tecmint.com/set-limits-on-user-processes-using-ulimit-in-linux/) are possible

rootzoll, 2018-08-27:
Remove the sudo from pi needs more work to keep info-scripts running - thinking about not investing the time in this for now, because maybe we will make jump to rewrite for ePaper Hat (replacing LCD) anyway soon. Will gather feedback on that from the community on the hackday this week.

But the exec sounds like a quick win. Will test it out. Much thanks.

rootzoll, 2018-08-27:
exec works great :D

closed 2018-12-11 by rootzoll