raspiblitz/raspiblitz #55: Supporting additional altcoins

chainfailure, 2018-08-24:
Are you guys accepting PR's that add additional altcoins? I'm currently working on a Viacoin raspiblitz, and was planning on maintaining my own fork. However I would much prefer not diverting from upstream like that.

rootzoll, 2018-08-24:
We already have litecoin integration. If its like litecoin a altcoin fork based on bitcoin source and will not add further complexity (just a third if-else here and there) why not. You can send PR along ... just might discuss it with the greater community before deciding adding.

I dont know Viacoin - maybe also not the other. Can you give a short introduction:
Are there binaries compiled for ARM (like for bitcoin and litecoin)?
How big is the Blockchain?
TOR seems a big deal on your website .. would you like to review upcoming TOR feature for raspiblitz and give feedback if all implemented legit?

rootzoll, 2018-12-24:
Still open for others alt-coins, but ticket is kind of old now. So I closing it for now.

closed 2018-12-24 by rootzoll