raspiblitz/raspiblitz #56: error after change of hostname

vv01f, 2018-08-25:
ref: https://github.com/rootzoll/raspiblitz/blob/77a8f90402cc3a17be98394854b546ec22c89376/build.sdcard/raspbianStretchDesktop.sh#L36

problem: when manually applying the script, this line results in error "unable to resolve host raspberrypi"

measure: applying hostname after other changes such as memory split

vv01f, 2018-08-25:
relating to such problems I'd like to suggest using `cmd1 && cmd2` in case `cmd2` depends on `cmd1` etc.

rootzoll, 2018-08-27:
Changing hostname should be already at end of the setup - right before the final reboot. Let me check for leftovers of old code.

rootzoll, 2018-08-27:
Found it in the manual SD card setup. Good find. Thanks. Will move to the end before reboot / LCD setup.

closed 2018-08-27 by rootzoll