raspiblitz/raspiblitz #61: enhance torrent

vv01f, 2018-08-27:
the torrent needs quite a while, mainly because it is seeded rarely; this could be changed if the torrent continues to seed later on after setup.

* [ ] switch from transmission-cli to transmission-daemon
* [ ] show download status in infoLCD
* [ ] create dir for torrents
 + [ ] watch-dir `/mnt/hdd/torrents`
 + [ ] download-dir `/mn/hdd/complete`
 + [ ] incomplete-dir `/mnt/hdd/incomplete`
* [ ] replace rpc-password with a random string, maybe customized
* [ ] add local ip range besides localhost to rpc-whiteliste
* [ ] ensure no download to sdcard to lower wear 

```~/.config/transmission-daemon/settings.json
{
    "alt-speed-down": 50,
    "alt-speed-enabled": false,
    "alt-speed-time-begin": 540,
    "alt-speed-time-day": 127,
    "alt-speed-time-enabled": false,
    "alt-speed-time-end": 1020,
    "alt-speed-up": 50,
    "bind-address-ipv4": "0.0.0.0",
    "bind-address-ipv6": "::",
    "blocklist-enabled": true,
    "blocklist-url": "http://john.bitsurge.net/public/biglist.p2p.gz",
    "cache-size-mb": 4,
    "dht-enabled": true,
    "download-dir": "/mnt/hdd/complete",
    "download-queue-enabled": true,
    "download-queue-size": 5,
    "encryption": 1,
    "idle-seeding-limit": 30,
    "idle-seeding-limit-enabled": false,
    "incomplete-dir": "/mnt/hdd/incomplete",
    "incomplete-dir-enabled": false,
    "lpd-enabled": false,
    "message-level": 2,
    "peer-congestion-algorithm": "",
    "peer-id-ttl-hours": 6,
    "peer-limit-global": 200,
    "peer-limit-per-torrent": 50,
    "peer-port": 51413,
    "peer-port-random-high": 65535,
    "peer-port-random-low": 49152,
    "peer-port-random-on-start": false,
    "peer-socket-tos": "default",
    "pex-enabled": true,
    "pidfile": "/mnt/hdd/torrents/d.pid",
    "port-forwarding-enabled": true,
    "preallocation": 1,
    "prefetch-enabled": true,
    "queue-stalled-enabled": true,
    "queue-stalled-minutes": 30,
    "ratio-limit": 2,
    "ratio-limit-enabled": false,
    "rename-partial-files": true,
    "rpc-authentication-required": false,
    "rpc-bind-address": "0.0.0.0",
    "rpc-enabled": true,
    "rpc-host-whitelist": "",
    "rpc-host-whitelist-enabled": false,
    "rpc-password": "{e4d4936caaec384fde6b035763c8c3b38dd51e1bUhgPOiJp",
    "rpc-port": 9091,
    "rpc-url": "/transmission/",
    "rpc-username": "",
    "rpc-whitelist": "127.0.0.1,10.0.0.*",
    "rpc-whitelist-enabled": true,
    "scrape-paused-torrents-enabled": true,
    "script-torrent-done-enabled": false,
    "script-torrent-done-filename": "",
    "seed-queue-enabled": false,
    "seed-queue-size": 10,
    "speed-limit-down": 100,
    "speed-limit-down-enabled": false,
    "speed-limit-up": 100,
    "speed-limit-up-enabled": false,
    "start-added-torrents": true,
    "trash-original-torrent-files": false,
    "umask": 18,
    "upload-slots-per-torrent": 14,
    "utp-enabled": true,
    "watch-dir": "/mnt/hdd/torrents/",
    "watch-dir-enabled": true
```

rootzoll, 2018-08-27:
Had some similar idea in mind. Never worked with transmission-daemon before, can you point to a good tutorial/guide for debian/rasbian?

vv01f, 2018-08-27:
woked here:

* just ran it as user admin
* adjusted settings
* created directories
* added torrent to watch-dir => automatically was started
* still waiting for it to finish while softlink to directory already is in place for user pi to continue

… but will test with a fresh raspbian to ensure it works in the script (like create dirs based on settings file) as well.

rootzoll, 2018-08-27:
Feel free to make a pullrequest containing the torrent-daemon integration when your tests work out. Can the daemon also subscribe to a RSS where we can publish updated torrents? Old torrents need then to be deleted at one point to free HDD space.

vv01f, 2018-08-27:
updated ticket on e.g. rss and deleting torrents

rootzoll, 2018-12-24:
torrents seem to run quite stable in the latest versions - if anyone has detailed instructions how to further improve, please add here .. closing this for now.

closed 2018-12-24 by rootzoll