raspiblitz/raspiblitz #62: backlight control for original 7" touch display

vv01f, 2018-08-27:
the files to control the backlight is owned by root after each reboot, on option would be to change rights at boot time

backlight on/off toggle:

```
#!/bin/sh
f=/sys/class/backlight/rpi_backlight/bl_power
b=$(( 1 - $(cat ${f}) ))
test ${b} -eq 0 && echo "switching on" || echo "switching off" 
sudo sh -c "echo ${b} > ${f}"
```

change brightness within range 0-255:

```
#!/bin/sh
test $# -eq 0 && { echo "missing parameter: brightness (0-255)" && exit 1; }
f=/sys/class/backlight/rpi_backlight/brightness
test $1 = "high" && b=255 && shift
test $1 = "on" && b=200 && shift
test $1 = "normal" && b=128 && shift
test $1 = "low" && b=25 && shift
test $1 = "off" && b=0 && shift
test $# -eq 1 && test $1 -ge 0 && test $1 -le 255 && b=$1 && shift
test $# -gt 0 && { echo "unknown parameter: $@ – try values 0 to 255" && exit 2; }
sudo sh -c "echo ${b} > ${f}"
```

rootzoll, 2018-08-27:
Thanks for the research. So we could add this option in the menu, the user could make new settings and then we offer a reboot, right?

vv01f, 2018-08-27:
no reboot needed, still have to find out where to add … but I'll do then as you ask.