raspiblitz/raspiblitz #63: error: couldn't connect to server: unknown (code -1)

tomski174, 2018-08-27:
Hi,
Could someone please help me understand why I'm getting this error while trying to switch to mainnet?
[image]

tomski174, 2018-08-27:
I followed the setup as described. Then I selected switch to mainnet from the menu and received the error 
Here's the log. I hope this is what you looking for. 

[image]

rootzoll, 2018-08-27:
@tomski174 could be that the download of the mainnet blockchain was faulty. Where you doing torrent or FTP download? 

The downloading the blockchain part is still experimental. First goal was to make the RaspiBlitz for Hackathon events and deliver Blockchain already prepared on the HDD. Next goal is to get the downloads right and solid. First report like this. We need to improve here .. so happy for every detailed report. Thx.

tomski174, 2018-08-27:
I used FTP download.  Are u saying my Blockchain download is not valid?

rootzoll, 2018-08-27:
@tomski174 Yes - thats my best guess. Index looks broken :( Bitcoin mainchain is served still from my FTP at home. Will upload today/tomorrow new version where the FTP will be served from better server. Hopefully this will prevent this. But we are still in the try/error phase on that topic.

tomski174, 2018-08-27:
Ok. Thank you. I will download the Blockchain again once i get the go ahead from you.  Is there any way you can increase the upload speed on your side? At the moment it takes 5 days to download.  Thanks.

vv01f, 2018-08-27:
@tomski174 took me 2d here, I'll leave the seed online, so hope that helps.

CarlosRoldanx, 2018-09-10:
Has this issue been solved? or is there any way to overcome it?

pettergo, 2018-09-11:
i'm getting something similar, i was updating the sd card from raspiblitz-2018-08-15. 

2018-09-11 14:15:16 Cache configuration:
2018-09-11 14:15:16 * Using 12.5MiB for block index database
2018-09-11 14:15:16 * Using 8.0MiB for chain state database
2018-09-11 14:15:16 * Using 79.5MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)
2018-09-11 14:15:16 init message: Loading block index...
2018-09-11 14:15:16 Opening LevelDB in /home/bitcoin/.bitcoin/blocks/index
2018-09-11 14:15:16 Corruption: 1 missing files; e.g.: /home/bitcoin/.bitcoin/blocks/index/128529.ldb
2018-09-11 14:15:16 Database corrupted
2018-09-11 14:15:16 : Error opening block database.
Please restart with -reindex or -reindex-chainstate to recover.
2018-09-11 14:15:16 Aborted block database rebuild. Exiting.
2018-09-11 14:15:16 Shutdown: In progress...
2018-09-11 14:15:16 scheduler thread interrupt
2018-09-11 14:15:16 Shutdown: done

should i go ahead and try
 -reindex bitcoind 
or  
-reindex-chainstate bitcoind

and see what happens?

rootzoll, 2018-09-21:
new sd card image I will upload today will have new torrent client - looks much more reliable. Lets keep fingers crossed.