raspiblitz/raspiblitz #64: update to LND v0.5

vnnkl, 2018-08-27:
0.5 was just released,
https://github.com/lightningnetwork/lnd/releases

which brings nice updates, sendToRoute to balance channels for example.
would make sense to install 0.5 from the beginning or include an update logic that enables to update to latest stable release from lnd github

rootzoll, 2018-08-28:
Yeah :) Updated the sd card build file. Will test and upload new sc card image soon.

vnnkl, 2018-08-28:
don't get too excited, it is RC1

"not full on release, just a version that will have no additional features other than bug fixes if nothing major is discovered we can have the new version sometime next week possibly we have a fix for something discovered today, so that’ll prob warrant an rc2, which will hold off on to see if we can package any additional fixes w/ it" - roasbeef in lnd slack

so the final version should be expected next week, just after lightning hackday

but we might already add an update logic similiar to
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_faq.md#how-to-upgrade-lnd
already using https://github.com/lightningnetwork/lnd/releases/latest, which should always download the latest stable release (not release candidates)

rootzoll, 2018-08-28:
Like with the move of the macaroon path recently .. just an update of lnd may break the raspiblitz shell scripts. So there are dependencies we have to take into account when updating. 

At the hackday I want to provide version 0.9 of raspiblitz with the lnd 0.5 RC1 and gather feedback and experiences directly on the overall setup. Then I will add all the contribs and change within the setup whats practical and make a version 1.0 with RC2 or even a final 0.5-beta realease of lnd. From this moment on, there should be some kind of update mechanism until we reach version 2.0

Until then for update you should: close all channels, move your funds out. cleanHDD. shutdown. flash SD card with new image. Rerun setup. Move funds back in. Reopen channels. ... I know its a bit time consuming and changes node address, etc .. but its also helping on retesting the new setup process.

closed 2018-08-28 by rootzoll

kccheung, 2018-09-15:
@rootzoll I attempted to update the lnd from v0.4.2 (with an old sdcard image) to v0.5-beta by following [this](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_faq.md#how-to-upgrade-lnd), but failed.

Is the suggested way to update still like the one you mentioned?
i.e. "close all channels, move your funds out. cleanHDD. shutdown. flash SD card with new image"