raspiblitz/raspiblitz #66: test/fix restoring an old lnd wallet

rootzoll, 2018-08-28:
On the lastest test of restoring a lnd wallet from cypher word seed on lnd wallet setup - the lnd gets during scanning the chain into some undefined state with porinting the following in the outputs ... still not sure why:

`2018-08-28 19:10:49.467 [WRN] LNWL: Could not parse output script in 850fc5627818fe8249b71a920247ca4f6b3c7706149c4d04a616047baba1631a:2: opcode OP_DATA_75 requires 76 bytes, but script only has 39 remaining`

Needs more testing and fixing. If anybody can repoert if this worked or getting the same error, that would be great.

rootzoll, 2018-08-29:
OK .. it took just a loong time. Not nice, but also not critial. Will close for now.

Smiggel, 2018-10-19:
I have this issue on hand here. I get the exact same kind of error, when restoring a wallet.

rootzoll, 2018-11-04:
reopening the issue.

@Smiggel can you provide more background on your this? I am still not sure where this is coming from. Would be great if someone can research this a bit more - maybe get in contact with LND on slack.

Smiggel, 2018-11-04:
@rootzoll, it looks like a lightning issue. The software keeps syncing and continuers without problem. Not sure what is causing the problem, but it does not affect how the node works.

rootzoll, 2018-12-24:
Havent seen this with latest lnd. So closing for now.

closed 2018-12-24 by rootzoll

MrManPew, 2020-10-29:
I also get these - @guggero ?

nipponese, 2021-09-08:
I am seeing this issue with 0.13.1-beta,

guggero, 2021-09-08:
You can ignore those errors. It's just some weird data in some of the early blocks that lnd doesn't know what to do with. We should probably silence those errors, yes. But they're produced deep down in the chain logic so a lot of dependency updates for a single warning message.