raspiblitz/raspiblitz #69: Remove txindex from bitcoin config

rootzoll, 2018-08-30:
LND does not need a index blockchain anymore (kind of missed that change). Check if its safe to remove (for other features) and get some workload off the RaspiBlitz.
https://github.com/lightningnetwork/lnd/pull/751

rootzoll, 2018-08-30:
On the first look c-lightning (for future integration) could be still in the need for a indexed blockchain. We have to take a closer look at this.

erkarl, 2018-09-06:
From the release notes of 0.5 RC2:
> With this version of lnd, running a full node backend with a transaction index is now optional! As a result, if a user wishes to run a lighter version of their full node without the transaction index, then they're able to do so. **However, for performance reasons until the persistent height hints are re-activated, we recommend running with an active transaction index for your full node which backs lnd.** In either case, lnd will automatically detect if the backing full node has an active transaction index and act accordingly.

drunkenclam, 2018-09-20:
Im running raspiblitz with a pruned non-indexed bitcoind backend since several weeks (but on SSD, not HDD). Everything's fine so far.

rootzoll, 2018-09-21:
@drunkenclam thx for sharing your experience

rootzoll, 2018-10-16:
I gave this a long pondering time. I decided to keep the txindex because the RaspiBlitz should be a fullnode with all basic services on. The RaspiBlitz should provide a all "features on" platform, so that all (future) services that might need a txindex will run. 

To make it more lightweight can be part of a follow up project or optional in the future.

closed 2018-10-16 by rootzoll

fluidvoice, 2019-03-17:
> The RaspiBlitz should provide a all "features on" platform, so that all (future) services that might need a txindex will run.

@rootzoll This is reasonable ***IF*** the costs associated with this are low.  Such as the cost to the user's time if it takes longer for IBD/syncing w/txindex, and potentially problems (ie index errors causing the script to reindex the DB).  **If it's not required NOW** but only for future services, perhaps it's better to just do a re-download or re-index of the blockchain when the time comes that it is actually required.

rootzoll, 2019-03-19:
Since the RaspiBolt also dropped txindex and I got feedback that even electrum server could be run without it I am open to drop txindex in a future release - see issue: https://github.com/rootzoll/raspiblitz/issues/227