raspiblitz/raspiblitz #70: License?

AnotherDroog, 2018-08-31:
Would like to integrate some of raspiblitz with our project, if the license is compatible.

vv01f, 2018-09-02:
May it would be helpful to emphasize your needs for compatibility. ;)
I would personally tend to AGPL 3+, businesses could argue they need to use proprietary stuff directly or dont like the hassle to do things right with freedom enforcing software; but maybe the project owner has different preferences.

AnotherDroog, 2018-09-03:
We’re MIT/BSD licensed and AGPL would not be acceptable there. We don’t mind having to share changes. 

I like your term, “freedom enforcing”. 

Are you one of the authors?

rootzoll, 2018-09-03:
Funny thing - I have totally forgot about the license until now.

Personally I like GPL idea of "enforcing freedom". But I looked at the licenses of lnd, c-lightning and bitcoin and they all do MIT/BSD (like) licenses. So I tend to keep it that way. 

@vv01f MIT/BSD should allow to change/fork to AGPL later on if needed, right?

AnotherDroog, 2018-09-04:
Danke! That’s cool. 

I believe you can always move to a more restrictive license later.

rootzoll, 2018-09-04:
I would like still to keep this discussion open for some days.

vv01f, 2018-09-06:
@rootzoll aye, X11/MIT (as [bitcoin, the code](https://github.com/bitcoin/bitcoin/blob/master/COPYING), itself) does not limit anyone to make a simple copy and reship with more restrictive license; that's for good and worse. Some people argue that MIT-like style gives more freedom as it does not restrict anyone.

Also possible to have different parts with different licenses, but I would not recommend this for easier handling.

Another way would be to go dual-licensing. Also needs manual handling, thus I personally do not like that.

For decision it depends if you have a strong opinion. If you want encourage people to go for it the fewer limits, the better.

jagottsicher, 2018-09-07:
I prefer a licence, which allows to build your own products from both. similar like your own linux distro. the most work is the script collection, or?

vv01f, 2018-09-07:
@jagottsicher even AGPL does not hinder you from buildung your products. Just you would need to provide the source code. It is rarely a problem because of licenses that do not allow combination because they prohibit sharing. The point is: Companies regard the code their private asset. In my eyes truth ist: The asset is customer relations and capabilities for services.

also see [nina's “Copyright is Brain Damage”](https://www.youtube.com/watch?v=XO9FKQAxWZc&feature=youtu.be&t=9) as a popular example outside of coding.

jagottsicher, 2018-09-08:
@Vv01f thanks for clarification. That is enlighting, but seems it is needed for me to read a lot and understand the licencening systems more.

rootzoll, 2018-09-13:
OK. I think everybody is cool with the MIT License for now - so I added a LICENSE file to the repo.

Feel free to report any use of this software that you think is not reflecting the spirit of the community and then I can start a process with contributers to see if a stricter OpenSource license would be a better fit then.

closed 2018-09-13 by rootzoll

nyxnor, 2021-06-18:
Reconsider MIT for GPL? Are there any implications for you? Rootzoll will be already notified, so ping @openoms @frennkie to give some thoughts about this please.

My short review:
MIT -> The uncomplicated license, easy to dealf if and is copyright.
GPL -> Freedom enforcing for derivatives, lots of clauses and is copyleft.

For historical purposes:
I personally think licenses are just means to fight for freedom when there is claim of owning an intellectual property, which is created by fiat laws. For the current time we leave in, free licenses protect the code from being claimed by other project and prohibiting further development from here.

rootzoll, 2021-06-19:
MIT is the uncomplicated  license and because we can just go one time MIT -> GPL and not reverse that I would tend to make that step only if there is a practical reason. Like for example:

1) If there would be a (hostile) forking project, that continues to take our code and closes it we can still change to GPL to cut that project off from exploiting further community contributions.

2) If there is an GPL dependency/library/fork that we wanna/need to use and would only be allowed if be accept that copyleft.

Especially the second case shows that it can make sense to wait - because I am not excatly shure about the GPL versioning and what you can mix up there. So say we changed already to a GPLv3 but then we wanna use a GPLv2 library or something. 

I am open for the discussion ... but thats at this point in time my view on that topic so far.