raspiblitz/raspiblitz #73: Download blockchain hangs on checksum error.

jetlagv1, 2018-09-02:
I've tried a few times now but when downloading the blockchain it hangs on a checksum error every time at the same block. See attached Pic of the console output attached. 
[image]

rootzoll, 2018-09-04:
Thanks for the report. This was the torrent dowload, right?

jetlagv1, 2018-09-04:
Yes, it was the torrent which is the only download option in the latest
version of raspiblitz at the moment.

jetlagv1, 2018-09-04:
> :
> Working on improving FTP/Torrent at the moment - it the moment its still
> too slow and unstable. But there should be two download options in the
> latest version >0.86: FTP or Torrent.

Hmm, I checked yesterday before posting this issue and I'm sure 2018-07-17b
was the version linked in the Readme. Anyway, I see there is a newer
version now dated 2018-08-28. Downloading it now. I'll let you know hoe it
goes. How was the hackday? 😎

rootzoll, 2018-09-04:
make sure to clean your HDD for a fresh start - just format it (FAT, etc) on another computer before retry.

the hackday was awesome :D still recovering ;)

rootzoll, 2018-09-04:
I will close this issue - because of old version. Reopen if you have still problems with it with the new version.

closed 2018-09-04 by rootzoll