raspiblitz/raspiblitz #75: NAT firewalled?

jagottsicher, 2018-09-03:
I will start several raspiblitz in China and need to know if they work correctly behind a NAT firewall. should work like any bitcoin/lightning node, or?

jagottsicher, 2018-09-04:
some background: 
Here in China you get a router for your what-ever-configured internet connection. but you get only access to a user account of the modem. so your router receives ALWAYS an IP from the Modem via DHCP. In your router you can use that signal and bridge it to another network. But your traffic will always routed through the modem and it's IP and you can not create port forwarding  for example. 

So you can for example NOT run your own server at home. 

Due to the fact my wallets on PC and mac work properly I guess this should not be an issue, but I am afraid that incoming requests to provide an invoice could fail, because they never reach the destination. 

Anybody experience with any kind of lightning full node in China at an internet connection providing dynamically IPs?

rootzoll, 2018-09-04:
No experience with running nodes Nodes in China on my side. But NAT is a common problem for users. If your node is not public reachable othe nodes cannot open a channel with you. But you can open a channel with other public nodes. So then you can start to make payments. Also your channel gets published on the gossip network and other nodes have your node in their network map and should be able to send payments to you (if you have receiving capacity on your channel). 

One trick to build up receiving capacity would be you have a public node outside the NAT that is your own. Then you start up your node behind the NAT and make a channel with your public node. If the channel is funded you sent all the funding to your public node (its stays in the family). Now you have the full receiving capacity of that node behind the NAT available and you can handout invoices.

All that said. Of course its the best if your node behind the NAT gets public available somehow. The RaspiBlitz has experimental TOR integration in the latest version. Even behind a NAT your node will then be publicly available over TOR. 

Other ways to get this done could be to experiment with ngrok, cjdns or IPv6/VPN tunnels.

jagottsicher, 2018-09-04:
Thanks you very much. I am operating several lightning nodes on vps already which i can establish channels to.

Due to the fact i would like to become a forwarding relay within china the suggested solution will probably only support that indirectly. I needed to open lots of channels (especially to other in china located) nodes. 

I am aiming to become an useful entrypoint within China. Harder than i thought, but tor will probably help.

We‘ll see.

rootzoll, 2018-09-04:
Good luck. Share your experience :) Please keep in mind, that the RaspiBlitz is still a hackathon style project. Running big funds in production on mainnet, can lead to lost or stolen funds.

closed 2018-09-04 by rootzoll