raspiblitz/raspiblitz #78: Public IP not reachable

tomski174, 2018-09-04:
Sorry for another noob question.  Finally after my 3rd try i seem to be making progress.  I have the info screen  show public ip and port not reachable in red. Could someone please explain what do i need to do to resolve this.  Thanks. 

[image]

rootzoll, 2018-09-04:
Congrats for gatting it running :)

Your node is probaly behind the NAT of your router. The router has the public IP .. but if someone would call this IP on this port it would end up at the router - not getting forwarded to the RaspiBlitz. Check PortForwarding of your router to change this and set port 8333 and 9735 to be forwarded to the RaspiBlitz (175.33.15.48).

More details what in means for a lightnign node to be behind a NAT you can see here:
https://github.com/rootzoll/raspiblitz/issues/75

hopitas, 2018-09-14:
It shows not reachable for me too, but checking from [https://bitnodes.earn.com/](url) my node is found. I have opened and forwarded the port...

rootzoll, 2018-09-15:
@hopitas the check if a node if public is done by:
public_check=$(timeout 2s nc -z ${public_ip} ${public_port}; echo $?)

not sure if this is the best way - works on my machine - it was in the script I used. I f you know a better way, feel free to share.

Head, 2018-10-22:
@hopitas @rootzoll same here. Behind a NAT and ports are forwarded.
nc on 8333 times out, other ports refused the connections immediately. Seems like a firewall issue to me.

rootzoll, 2018-11-04:
I reopen to recheck that again for version 1.0

rootzoll, 2018-12-12:
For the moment I didnt found a better way then using netcat/nc command to test if ports are open. I upped the timeout from 2secs to 6 secs. 

I removed the "green" color because netcat port scan can just only test if there is a port open on that router - not that it is really pointing to that raspiblitz within the local network behind the router - could be another service or device at that port. It only uses the color "red" if there is no port listening at the router at all.

In the future it would be cool to improve the port analysis. For version 1.0 I will keep it at that level.

closed 2018-12-12 by rootzoll