raspiblitz/raspiblitz #79: Wallet backup service HDD->SD

rootzoll, 2018-09-05:
see https://twitter.com/JohnDoeCoin/status/1037237581199671301

drunkenclam, 2018-09-05:
I'm not sure but there still might be a problem if you reimport a backup with old channel states where you could lose your balance in the channel ("justice will be served").

And "old" backup could mean something like 2 minutes old, because you routed a payment 1 minute ago and your channel balances changed...

rootzoll, 2018-12-24:
Here are some more details about a backup solution for LND.
https://gist.github.com/bretton/22f628caffde79390a796e75ea528053

Automatic backup would need stop/restart of LND. Newest RaspiBlitz has auto-unlock feature, but this is a too complicated topic for now - closing the ticket. Its better to see if any features from LND come up in the near future and run with those.

closed 2018-12-24 by rootzoll