raspiblitz/raspiblitz #83: make touch/pen usable on cli “UI”

vv01f, 2018-09-06:
source: https://www.raspberrypi.org/forums/viewtopic.php?p=839857

It would be a better usability if the pen/touch displays could be used like a simple POS without an ssh connection or full keyboard.

rootzoll, 2018-09-08:
That would one way to go, but maybe not the best time investment. 

Because 1) i would like to change the LCD in the future to a ePaper HAT like this (no touch, but buttons) and 2) POS should ne done thru a connected mobile device (like smartphone) - much better and secure.

vv01f, 2018-09-12:
ok, not reasonable for the assumed usecase

closed 2018-09-12 by vv01f

rootzoll, 2019-02-22:
Because this was asked again on twitter:
https://twitter.com/raumi75/status/1099066019899953152

Some more resources if someone likes to dig deeper here and activate touch ... here is the manual of the LCD: https://www.elegoo.com/tutorial/Elegoo%203.5%20inch%20Touch%20Screen%20User%20Manual%20V1.00.2017.10.09.zip