raspiblitz/raspiblitz #84: Running through TOR

tomski174, 2018-09-09:
Are there any limitations to my node if i choose to run it through TOR?

rootzoll, 2018-09-09:
Ther are two things when behind TOR: 

1) LND RPC (Port 10009) : Not much Mobile wallets or other services that making connection to LND RPC have TOR support (yet). Sometimes its possible you can use bridge thru a local TOR proxy but thats very tricky.

2) LND Node (Port 9735):  Other nodes that dont have running TOR cannot connect to you. But you should be able to open channels to non-TOR nodes.

This is at least based on my experience with TOR when trying it out. Would love to hear other experiences runnign a node behind TOR.

rootzoll, 2018-09-15:
@tomski174 thats cool seeing better support for TOR inbound connections. I already preparing new LND 0.5 release for the next sd card image - hopefully being ready for next week.

rootzoll, 2018-10-03:
TODO: recheck TOR with new setup

rootzoll, 2018-12-05:
OK TOR is now part of the service section. Install and Deinstall look good on tests. Closing Ticket for now.

closed 2018-12-05 by rootzoll