raspiblitz/raspiblitz #85: Slow Download via torrent /FTP hangs all the time

jagottsicher, 2018-09-09:
I tried both the torrent (which is not well distributed) and the ftp-download, which hangs after a few minutes.

After that I tried to download via an ordinary FTP-client, but after logging anonymously. all the time timeouts and not possible to download a single file. The torrent took me with vuze about 24 hours. 

Here a solution which worked: 
Download the torrent. add as many trackers as possible. copy the 220G to a linux machine in the same network like the raspiblitz(e) are located in. the rsync the data from taht machine, after rebooting, but NOT coninueing the setup. 

On the source linux machine use

`rsync -av --delete --progress /source/downloads/raspiblitz-bitcoin-2018-07-16/  -e "ssh" admin@IP_OF_YOUR_RASPIBLITZ:/mnt/hdd/raspiblitz-bitcoin-2018-07-16/
`
That should do the job. Please wait untl the download is completed. 

No idea which user and permissions you need, but after rsyncing to the destination open a shell there and simply type 

`chown -R admin:admin /mnt/hdd/raspiblitz-bitcoin-2018-07-16` 

It should work here. not sure about the group, feel free to correct me.

jagottsicher, 2018-09-09:
After double-checking I saw that the directories' owner is root:root and the permissen settings are 755.
Due to the fact you can not ssh in the machines as root, hence you can not rsync as root, you need TEMPORARELY to set the permissions to 777 for Rsync.

DON'T forget to set back to 755, or even develop some stricter policy.

jagottsicher, 2018-09-10:
saw your postings that you're busy.

for those who downloaded the torrent and don't know how to proceed:

If you can ensure you downloaded the folder 'raspiblitz-bitcoin-2018-07-16" and its content then you can proceed "as you copied from another disk. That means you first should set the permissions for that folder to 644.

Then you need to call the script  `60finishHDD.sh`, but as you need higher permissions you can just login as "admin" and `sudo ./ 60finishHDD.sh` 

the script will pepare the disk (set permissions and owner), start bitcoind and eventually start lnd 

Not sure, but it shoudl also word to reboot after renaming the folder.

jagottsicher, 2018-09-10:
obviously after restart the script tangle up somewhere and asked passwords again.

Make sure that data in /home/admin/.bitcoin/bitcoin.conf and /home/bitcoin/.bitcoin/bitcoin.conf are correct (rpcuser and rpcpassword) and add the data if needed to /home/admin/.lnd/lnd.conf

Else you have no access with bitcoin-cli

vmavromatis, 2018-09-14:
Any list of trackers or any other help? I'm running for 3 days non-stop, only at 48.6% :( FTP hangs all the time. Any help would be much appreciated.

rootzoll, 2018-09-14:
I am testing new torrent client and new ftp server at the moment. Hope to
get new sd card image out next week.

2018, 17:00:

jagottsicher, 2018-09-15:
[image]

Sorry , only as pic.

Btw: downloading the blockchain is terrible, but the worst to come: lnd needs days scanning the blockchain.

vmavromatis, 2018-09-15:
Thank you @jagottsicher added. Another feature I'm looking forward is being able to run this with txindex=0. I'm also wondering why are we not a more public version of the torrent such as the one on https://getbitcoinblockchain.com/ but maybe I'm missing something.

rootzoll, 2018-09-15:
@vmavromatis I didnt knew about: getbitcoinblockchain.com - it would be great to use an already good seeded blockchain that is not older than a month. 

The blockchain they provide is not indexed (txindex=0), correct?

If the raspiblitz schould/could work with an unindexed blockchain is still an open issue: https://github.com/rootzoll/raspiblitz/issues/69 @vmavromatis would be great if you can share your experience to help us making a decision.

rootzoll, 2018-09-15:
Update on the progress: 

Torrent -> I changed bittorrent client to rtorrent - looks good so far. Starts downloading much faster/quicker than transmission-cli and also does not abort when get started.

FTP -> Tested new FTP server. Did not shutdown/throttle during my tests. But the data downloaded had a checksum error reported by bitcoind. I think I heard this error before with the old server - so it may be the data that this on the server. I prepare to update bitcoin blockchain data on FTP at the moment and have to run second test.

rootzoll, 2018-09-15:
@jagottsicher if I make another/new torrent what trackers would you recommend. The ones that are showing success in your image?

jagottsicher, 2018-09-17:
> 
> 
> @jagottsicher if I make another/new torrent what trackers would you recommend. The ones that are showing success in your image?

I can only speak for China. The aktive trackers worked. The inactive one I never could reach.

rootzoll, 2018-10-18:
Created new torrents (it now two: base and update). See background info at:
https://github.com/rootzoll/raspiblitz/blob/master/background_downloadBlockchain.md

I hope this should improve the torrent seeding. Closing issue for know - reopen, if with version 0.95 sd card image its still an issue.

closed 2018-10-18 by rootzoll