raspiblitz/raspiblitz #88: infoBlitz with IPv6

frennkie, 2018-09-10:
I have IPv6 on my local network and the infoBlitz Screen does not look pretty:

```
              RaspiBlitz v0.86  raspiblitz
              bitcoin Fullnode + Lightning Network + TOR
        ,/    -------------------------------------------
      ,'/     load average: 1.66, 2.12, 1.36, CPU 61°C
    ,' /      Free Mem 561M / 927M  Free HDD 642G (70%)
  ,'  /_____, ssh admin@192.168.178.116 ▼17.1MiB ▲47.2MiB
 .'____    ,'
      /  ,'   bitcoin v0.16.2 mainnet Sync OK (100.00%)
     / ,'     Public 2001:XXXX:XXXX:XXXX:XXXX:XXXX:XXX:XX59
2001:XXXX:XXXX:XXXX:XXXX:XXXX:XXXX:XX37:8333
    /,'
   /'         LND v0.5-beta wallet 124575 sat
              2/3 Channels 227533 sat

03730a3bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxd9a@123.123.123.123:9735
```

When I have some more I'll try to create a PR for this.

frennkie, 2018-09-10:
Also - I have a 3,5" LCD running at `480X320` and it cust of the last digit of the `Public 2001...` line.

frennkie, 2018-09-10:
Hm.. the following line finds my non Onion/Tor address and then thinks I use Tor (which I don't):
```
onionAddress=$(${network}-cli -datadir=${bitcoin_dir} getnetworkinfo | grep '"address"' | cut -d '"' -f4)
``` 

Maybe it would be "safer"/better to parse the JSON return by `getnetworkinfo` and check whether `["networks"]["name"] == "onion"` and `["networks"]["reachable"] == true` for the Onion/Tor check..?!

rootzoll, 2018-09-21:
I could be that in the future the display info will be managed with another programming language - parsen JSON would be the better available option then.

frennkie, 2018-09-21:
Personally I have to say I do like the current *old school* style of the info screen. But I also think it's currently limited regarding functionality.

I think there would be two alternative approaches:

* a GUI interface (e. G. Python/QT or sth else) 
* a HTTP based backend and a small Browser opening it

Both should enable us to use the touch screen, e. g. for switching view or generating and displaying an invoice.

What do you think?

rootzoll, 2018-09-21:
For making it more clear - there are tweo UI parts at the moment: There is the LCD and the SSH Terminal Dialogs. SSH works for technical people - but not for everybody - so the idea is that you dont open the terminal anymore in the future - you connect with a webbrowser within the local network - like with your router.

The other thing is the LCD screen. Here for the future there is the idea to switch to a ePaper - at least as the preferred option. And because the ePaper cannot display info like a monitor and needs to be managed in a different way and with that JSON parsing might come.

If anybody likes to invest time to evolve also the LCD parrallel to the ePaper I am happy to coordinate.

frennkie, 2018-10-02:
@rootzoll I re-implemented the LCD Info Screen (`00infoBlitz.sh`) in Python (`00infoBlitz.sh`) . I believe this will give more flexibility - **especially** when errors/timeouts need to be caught.

I'm not completely finished and I wasn't able to test every possible screen content. But at least I fixed the broken layout for my IPv4+IPv6 (but no Tor) setup:

[image]

What do you think? https://github.com/frennkie/raspiblitz/tree/rewrite-infoblitz

What would it mean to adapt this for an ePaper screen?

rootzoll, 2018-10-03:
@frennkie wow thats cool - will check it out over the next days, just back from vacation.

ePaper will need extra library - but demo code is also in python on the waveshare wiki:
https://www.waveshare.com/wiki/2.7inch_e-Paper_HAT_(B) 
So there is a good chance that you can port this to ePaper :D

frennkie, 2018-10-06:
@rootzoll I have put some more work into my Python version of the InfoBlitz dashboard. I think every metric is now migrated and I try to handle all errors. So if something goes wrong the dashboard should still keep showing and the errors/warnings should only go to a log file (infoblitz.log) for further debugging/inspection.

frennkie, 2018-10-12:
@rootzoll Did you have a chance to give this a test run...? Does it make sense for me to continue working on this?

rootzoll, 2018-11-04:
It took a while but now lets pick up on this topic. Help me on walk thru your work:

I can find the script here:
https://github.com/frennkie/raspiblitz/tree/rewrite-infoblitz

There is also a __init__.py .. is it needed? I would like to move the py-script to /home.admin/00infoBlitz.py ... so that its easy for people to switch between shell and py is they like.

rootzoll, 2018-12-24:
For the version 1.0 I stuck to the old shell script to enhance the displaying - but the py-version will be a good base to go towards a display service in the future.

rootzoll, 2019-02-22:
About a future "display service" it might be worth looking at the now OpenSource code of the Casa Node. For their WebGUI they should have something similar.

raumi75, 2019-02-26:
A small browser on the LCD is surprisingly light weight. After watching this [35C3 Talk](https://media.ccc.de/v/35c3chaoswest-48-wie-aus-einer-wette-ein-kiosk-system-fr-den-raspi-wurde), I managed to get the epiphany and the chromium browser to run.

#!/bin/sh
```bash
#!/bin/sh
xset -dpms # disable DPMS (Energy Star) features.
xset s off # disable screen saver
xset s noblank # don't blank the video device
xset -nocursor
#matchbox-window-manager -use_titlebar no &
# prequisit mkdir /home/admin/.config
WEBKIT_DISABLE_TBS=1 epiphany-browser -a --profile /home/admin/.config file:///home/admin/status.html
```

and ZZstartBrowser.sh

```bash
#!/bin/sh
sudo xinit /home/admin/ZZstartBrowser.sh &
```

My proof of concept website looks like this:

status.html
```html
<!DOCTYPE HTML>
<html>

  <head>
    <meta http-equiv="Content-Type" content="text/html; charset=UTF-8" />
    <title>RaspiBlitz Proof of concept status page</title>
    <meta http-equiv="refresh" content="5" >
  </head>

  <body style="background-color:lightgrey;">

    <h1>raumiblitz</h1>

<p>bitcoin Fullnode + Lightning Network </p>
<p>, CPU °C</p>
<p>Free Mem 376M / 927M Free HDD </p>
<p>ssh admin@192.168.178.42 ▼31.6MiB ▲51.5MiB</p>
<p>web admin --> http://192.168.178.42:3000</p>
<p>bitcoin v0.17.0.1 mainnet Sync OK ()</p>
<p>DynDNS raumi.ignorelist.com 16 connections</p>
<p>LND 0.5.2-beta wallet 212784 sat </p>
<p>5/6 Channels 358874 sat 10 peers</p>
<img src="file:///home/admin/assets/install_shango.jpg" alt="qr-code" style="width:auto; height:8em;">
  </body>
</html>
```

the `http-equiv` makes it refresh every 5 seconds.

And now the fun part: I modified `00mainInfo.sh` to generate the status.html. I know that is extremely quick and dirty, but without JavaScript Bloat, it does not eat up valuable cpu cycles. We can style it with some CSS, maybe use nice fonts, etc.

I've never used python for creating a website. Have some experience with php, ruby and JavaScript. How hard can it be? What do you think, @frennkie and @rootzoll do you think this is a path worth going? 

[image]

rootzoll, 2019-02-26:
This looks super cool :D This week I am finishing up some other issues and next week I am on vacation ... but I promise I will get back on this and will help on this once I am back - this is awesome :D

rootzoll, 2019-02-26:
@raumi75 maybe you can connect with @frennkie - there is some ground work for a info-service the display/browser can then request the data from. 

Does the browser mode support touch? Because then we could emulate four (touch-)buttons on the left side and keep the basic layout and interactions on the LCD the same as on this ePaper Hat:
https://www.amazon.de/HAT-Resolution-Electronic-Controller-Communicating/dp/B075FQKSZ9

So there would be one "info-service" (local REST API) gathering data about the raspiblitz and either the LCD Browser or the ePaper-Controller can use this to display.

raumi75, 2019-02-27:
I've spent hours trying to get touch to work. I'm really wondering if my display is broken, because even when I start an x-session, I can't use it. Not even when I run the calibration. Very frustrating. When I connect a mouse, the pointer moves.

Has anyone gotten the touch to work?

The ePaper looks awesome, but have you seen how slowly it updates?

https://youtu.be/c_j6QWyRJLg?t=109

The video was kind of disappointing. This doesn't make much sense with fluctuating data like workload or network speed and temperature, right?

I was hoping to build some fun stuff like an animation when an invoice got payed or funds arrive, etc. That's nothing for ePaper. We'd have to come up with a nice looking static image for that.

Yes, @frennkie we should get together. There should be an API that can be used for displaying the information via ssh, LCD, ePaper and maybe even web service.

What 4 functions do you have in mind? I think power off and display QR for funding would be safe to have.

raumi75, 2019-02-27:
How do you feel about the hardware and the future of this project? We all know a pi is the bare minimum to make this run and I understand why people want a Rock64 and even an Odroid. Since they don't have a display, what do you think? How much should we emphasize this?

I for one think the LCD makes this "black box" very approachable. I remember how people stopped at the 35C3 assembly and looked at all those bright boxes on the table.

rootzoll, 2019-02-27:
Update Times of ePaper: The ePaper should be OK for partial updates. Some data would just be updated in longer periods. I think thats even OK for workload and temp.

4 Buttons: Still have to figure out what would be best here. But if the first button lets you circle thru a number of different screens the other 3 buttons can have flexible action each different to each screen. So we can start small with for example two screens and extend later what people most demand and find usefull.

Hardware: A RaspberryPi4 is not expected to be released this year. So yeah we have to see. I have a Rock64 and it has also those GPIO pins - if it would possible to get the LCD/ePaper also there running then that would be a good alternative option for people that want to go with more power. This field is open for experimentation.

LCD/ePaper: When I was alone working on this project I was tending more to just replace the LCD with the ePaper at one point to keep low of complexity and time resources. Now that the comitment of the community has grown I think both options could be build in parallel. The LCD as a default (because yes, it a great eye catcher :) and the ePaper as a replacement option if you want it power reduced and dont like the light emitting in your home.

frennkie, 2019-03-09:
@raumi75 @rootzoll Sorry for being late on reacting to this.. too many other side projects.. ;-)

I updated the initial comment of this issue to reflect more what this issue is now about.

> Update Times of ePaper: The ePaper should be OK for partial updates. Some data would just be updated in longer periods. I think thats even OK for workload and temp.

I agree.

> 4 Buttons

Yeah I am also not sure.. In the end it depends whether the Raspiblitz is somewhere in the back(office)/under the counter or whether you would actually but it "user facing" (which is one my uses cases - so then I would say "generate an invoice" would be at least one functions).

I really like the html status page (both light and dark) you created. It's also great to hear that this actually does not consume too many resources. This was a concern (my Chrome currently eats 2GB RAM for a few tab.. /o\ ).

As I indicated in the opening comment there should be a backend service that has the job to collect the metrics that are relevant for the status page. The browser then either reloads a static html page that is regularly re-generated or the browser queries a (REST) API to get the latest data.

**Backend**

I migrated the shell code that runs the current info display to Python: https://github.com/frennkie/raspiblitz/blob/rewrite-infoblitz2/home.admin/infoblitzd.py

**Please be advised**: This is far from finished/stable. This is the first time I implemented threads in Python and - well - I have encountered dead locks. These issues are not fixed. So I would advise against taking this over completely. But most of the "data gathering" functions should actually be fine.

In the mean time I thought about switching to a really small - off-the-shelf - database. Either on disc or in memory. My python code simply feeds into it and the frontends query it directly.

frennkie, 2019-03-10:
I created a rough architect drawing. Everything is open for discussion.

[image]

rootzoll, 2019-03-10:
@frennkie great overview :) I think a info gathering service is the way to go for the future - and having a REST API gives the most flexibility onhow to use it (ePaper, LCD, browser, RaspiBlitz-cli).

On which HTTP server to choose ... I think we should use the most resource saving but with having the felxibility to forward pathes to other services runnign local on ports. Like mapping a path "/RTL" to the RTL node server (when use has it activated).

frennkie, 2019-03-10:
@rootzoll 

> and having a REST API gives the most flexibility onhow to use it (ePaper, LCD, browser, RaspiBlitz-cli).

Fully agree. I have started looking into Redis. So far this looks great:
* very lightweight
* data is kept in memory - this should enable really low delay even if the Pi is busy with IO (chain scan)
* extremely broad library support (CLI, Python, JavaScript etc...)

Everything running on localhost (ePaper, LCD, local browser, RaspiBlitz-cli) can directly access Redis with one of the libs. Only if remote clients want to access this data an "application server" (e.g. NodeJS) or Webdis would be needed.

The redis that comes with the Raspbian sources is a bit outdated (3:3.2.6-3+deb9u1). Any concerns installing a more recent version from the backports (5:5.0.3-3~bpo9+2)?

This would be the instructions:

```
gpg --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 7638D0442B90D010 8B48AD6246925553
gpg --armor --export 7638D0442B90D010 | sudo apt-key add -
gpg --armor --export 8B48AD6246925553 | sudo apt-key add -

echo "# debian strech backports -> https://wiki.debian.org/Backports
deb http://deb.debian.org/debian stretch-backports main" | sudo tee -a /etc/apt/sources.list.d/raspi-backports.list

sudo apt-get update

# backports have lower priority in APT and therefore stretch-backports must be explicitly set
sudo apt-get install -t stretch-backports -s redis-server
```

> On which HTTP server to choose ... I think we should use the most resource saving but with having the felxibility to forward pathes to other services runnign local on ports.

I guess I would go with **nginx**. Also hugely popular, very flexible and supposedly pretty efficient. Especially for doing static file serving and - if needed - handling TLS for remote clients.

> Like mapping a path "/RTL" to the RTL node server (when user has it activated).

Yeah, I noticed that RTL builds on NodeJS and there is already an install option that installs the node binary. This is one reason why I'm also considering this as a part of the "core" service. But I'm not sure how "lightweight" that actually is.. do you experience with this?

frennkie, 2019-03-10:
Ok.. I'm in the middle of the rewrite.. but basically this already works.

#### Install redis

As per instructions above. There might be some setup required.. I used the settings from here: https://habilisbest.com/install-redis-on-your-raspberrypi

#### Install Python lib

`sudo -H pip3 install redis`

#### Get Work-in-Progress code

```
git clone https://github.com/frennkie/raspiblitz.git
cd raspiblitz
git checkout infoblitzd-redis
cd home.admin
./infoblitzd.py   # or python3 infoblitzd.py
```

#### Then in another shell (make sure NodeJS is installed)

```
sudo ufw allow proto tcp from 192.168.178.0/24 to any port 3000 comment 'allow local LAN to NodeJS'
cd raspiblitz/home.admin/service/
node app.js
```

Open Browser to http://raspiblitz:3000/

raumi75, 2019-03-11:
@frennkie Great to see you are making progress. I'll need to take a good look at this as soon as I have a few hours of time.

Just a quick heads-up: I believe port 3000 is used by RTL, also.

frennkie, 2019-03-11:
> Just a quick heads-up: I believe port 3000 is used by RTL, also.

Yeah.. this appears to be the default for NodeJS Express. I'm actually not so happy with Express... it is known to really "heavy".. using this was just really quick'n'dirty... :-)

@rootzoll: Any objections to spinning off this Display/Info Service into a separate repo? I would also publish this as a package of PyPI. This makes independent development and updating easier (pip install -U ).

fluidvoice, 2019-03-11:
It's a shame more people don't know the Elixir language. It can do all this and less complex architecture.

frennkie, 2019-03-11:
@fluidvoice I agree that some languages and paradigms are better suited than others.

For me Python is a "go-to" language.. and I actually already implemented almost all of what is in the picture in a "monolithic" python script. While you call this a "a complex architecture" I would at least add that it is flexible and "pluggable".

But I'm always open for improvement suggestions.

fluidvoice, 2019-03-11:
> @fluidvoice I agree that some languages and paradigms are better suited than others.
>
> For me Python is a "go-to" language.. and I actually already implemented almost all of what is in the picture in [...]

What do you think will be the memory and disk footprint, more or less?  I'm not so concerned about the complexity of all the pieces that need to be installed as how resource efficient it will be on the modest Pi device.

frennkie, 2019-03-11:
@fluidvoice Oh.. wow.. am I glad that you made me check the memory footprint..! You may have saved a lot of wasted time..

Just for the record here: My approach was to schedule separate tasks (each in a Python thread) for each Metric (the values that should be available for displaying). Python reserves 8MB for each thread. Given that I currently have 53 Metrics.. that would have been 400MB RAM.. ;-D

Ok.. so this definitely was the wrong approach... but of course there are others...

frennkie, 2019-03-18:
@rootzoll @raumi75 I have started working on an alternative approach that would be "self-contained" and would therefore not need additional services. I believe this also will enable a really "small footprint". This goes along the lines of @fluidvoice's suggestion .. but I chose **Golang** over **Elexir**.

https://github.com/frennkie/blitzinfod/releases/tag/v0.3-poc

frennkie, 2019-06-02:
Short update: The work on **blitzd** is continuing. I had some good discussion regarding this at the Hackday. The brief summary is this:

* I'm planning on building two binaries (blitzd and blitz-cli).
* Due to the nature of Golang there will be binaries for different platforms (arm, x86, amd64)
* There will be three separate servers:
  * An unencrypted HTTP (no certificate issues) "welcome" server -> this won't show any information the RaspiBlitz that would raise privacy/confidentiality concerns. Mostly it will have a brief intro and links to this Github page
  * An encrypted HTTPS "secure" server.. this will host @raumi75 's "HTML Status Page" so that users can remotely check the status (without SSH) - It will use BasicAuth (username + password) authentication.
  * A gRPC API server. This will - at first - only contain "read-only" access to certain metrics. But it definitely makes sense to also implement a few real services (most prominently a "controlled shutdown" service endpoint). Authentication is planned to be done with mutual TLS (client certificates) .. the alternative would be to look into macaroons..

frennkie, 2020-05-17:
As per our (@openoms @rootzoll) discussion I have started on a PR for adding nginx as a core part of RaspiBlitz: #1189.

The main idea's are similar to what is already outlined in the this thread here.

Nginx will host a plain information page on HTTP Port 80.

Nginx will host a HTTPS site on Port 443 that will include some kind of dashboard (at the beginning most likely similar to the LCD info screen).

Nginx will also act as a TLS Reverse proxy to provide access to several apps (e.g. LNbits, Btcpay, RTL). This will at first be done on a dedicated port each (e.g. LNbits on http 127.0.0.1:5000 and Nginx on https 0.0.0.0:5001 ..).

Nginx might be able to use SNI to provide access to several apps over port 443 (e.g. lnbits.raspiblitz.mydomain.com, rtl.raspiblitz.mydomain.com) later.

NilByte, 2020-12-19:
How about throwing this in somewhere?

`curl -s http://api.coindesk.com/v1/bpi/currentprice.json | python -c "import json, sys; print(json.load(sys.stdin)['bpi']['USD']['rate'])"`

rootzoll, 2020-12-24:
@NilByte this can make sense for a LCD screen plugin showing the price

rootzoll, 2022-03-09:
This is now done with the `background.scan.service`in v1.7.2 - the values can be requested thru the `/home/admin/_cache.sh` script. Closing issue.

closed 2022-03-09 by rootzoll