raspiblitz/raspiblitz #91: Error Code 28

toxicsh0ck, 2018-09-12:
sudo: unable to resolve host "USERNAME"
error code: -28
error message:
Rewinding blocks...
error code: -28
error message:
Rewinding blocks...
sudo: unable to resolve host "USERNAME"

loading block index...
"Unable to resolve host"

Any ideas?

rootzoll, 2018-09-13:
At which point in theprocess is that happening?

toxicsh0ck, 2018-09-13:
Its happening after the process has been completed and I am seeing the LND main screen, and then the error text scrolls down in orange.

Would this have anything to do with "Public IP” being in red and saying “unreachable”?  This IP Is not my devices IP.

Thanks in advance and for your work on this, and appreciate your help.  Not super tech savvy and the article was really helpful.

CarlosRoldanx, 2018-09-15:
Do not worry. Error code 28 means block synced. It should automatically load and connect to your blitz

vmavromatis, 2018-09-16:
Same issue happens to me (copied the blockchain from my hdd). Main screen says:
```
                                                                              ┌────────────────── bitcoin /  ─────────────────────┐
                                                                              │  Lightning Rescanning Blockchain                  │  
                                                                              │  Progress: ?/                                     │  
                                                                              │  Please wait - this can take some long time.      │  
                                                                              │  Its OK to close terminal and ssh back in later.  │  
                                                                              └───────────────────────────────────────────────────┘  
```
@CarlosRoldanx @rootzoll I suppose I should just wait?

CarlosRoldanx, 2018-09-16:
Yes, make sure you're connected to the internet and try the funding option. 

If you're not synced he wallet will not be created and will display the same error. 

Try again after X amount of time. If this time the wallet is created blocks are synced and you're connected

vmavromatis, 2018-09-16:
Update: Restarted and now getting the following
```
****************************************************************************
Unlock LND Wallet --> lncli --chain=bitcoin unlock
****************************************************************************
HELP: Enter your PASSWORD C
****************************************************************************
error: couldn't connect to server: unknown (code -1)
(make sure server is running and you are connecting to the correct RPC port)
Input wallet password: 
```
Funding error:
```
[lncli] rpc error: code = Unavailable desc = grpc: the connection is unavailable
FAIL PRECHECK - lncli getinfo shows 'synced_to_chain': false - wait until chain is sync 

******************************
TODO
******************************
Send REAL Bitcoin to address --> 
Whats next? --> Wait for confirmations. You can use lnbalance for main menu or info on LCD to check if funds have arrived.

Press ENTER to return to main menu.
^Cadmin@vasilightning:~ $ lncli --chain=bitcoin newaddress np2wkh
[lncli] rpc error: code = Unavailable desc = grpc: the connection is unavailable
```
After another restart:
```
****************************************************************************
error: couldn't connect to server: unknown (code -1)
(make sure server is running and you are connecting to the correct RPC port)
Input wallet password: 
[lncli] rpc error: code = Unavailable desc = grpc: the connection is unavailable
```

CarlosRoldanx, 2018-09-16:
I think it is related with either connection to the Internet.
Or syncing issues with LND

toxicsh0ck, 2018-09-16:
Great, thanks for the feedback.  If its a syncing issue with LND, any suggestions on how to remedy?

If its an internet connection issue, I assume this would be within the Router settings.  Can you guys please confirm what port LND runs on so I can ensure the device is set up appropriately?

Thank you.

CarlosRoldanx, 2018-09-17:
9735
But normally doesn't need a manual correction. You should be able to sync automatically, if you need a quicker help, ping me on telegram. I'm @whiteyhat

rootzoll, 2018-09-21:
Just a side note: Error Code 28 can also be displayed while bitcoind is doing some migration work or catchup/verifing. Normally this just takes time ... but if anybody has still a permanent Error Code 28 even with the latest version sd card image (2018-09-22) please write your report here. I will check after I get back from vacation.

svorob, 2018-09-23:
Here's what I am getting during set up after upgrading to the latest image:

```
OK - bitcoind is running
Next step run Lightning

*** Checking bitcoin ***
OK - bitcoind is running

*** Test if chaninnetwork is ready ...
1537720422
result()
error(error code: -28
error message:
Verifying blocks...)
2018-09-23T16:32:13Z Opening LevelDB in /home/bitcoin/.bitcoin/indexes/txindex
2018-09-23T16:32:13Z Opened LevelDB successfully
2018-09-23T16:32:13Z Using obfuscation key for /home/bitcoin/.bitcoin/indexes/txindex: 0000000000000000
2018-09-23T16:32:13Z Upgrading txindex database... [0%]
2018-09-23T16:32:13Z Upgrading txindex database... [39%]
Waiting 1 minute and then trying again ...
```

and goes on indefinitely without actually making any progress. However, every time I restart, the second number gets a little bit larger. Would appreciate any help, thank you.

machasm1, 2018-10-02:
This error is caused by the bitcoind still catching up or restarting.
In particular if you have copied the blockchain from another devide WITHOUT txindex enabled then the blockchain will need to be rebuilt which takes a loooong time (Days or even weeks if you are trying to rebuild on the pi).
Make sure you have a copy of the blockchain which has been synced using the txindex set. I had this problem myself and ended up firing up bitcoin-qt on my main PC which has a much more powerful CPU and RAM and then re-indexed the blockchain on that machine using the txindex option.
Once completed I then copied the chainstate and index folders to the raspibolt in the /mnt/hdd/bitcoin directory (chainstate) and /mnt/hdd/bitcoin/blocks directory for (Index).
After that it all started working just fine.

rootzoll, 2018-10-03:
The update of torrent and FTP (issue #127) should fix/reduce this for next release.

rootzoll, 2018-10-18:
With the transition to bitcoin 0.17.0 and the new blockchain downloads, this should not be a an issue anymore. closing.

closed 2018-10-18 by rootzoll