raspiblitz/raspiblitz #93: Mobile Wallet externalIP/domain

rootzoll, 2018-09-13:
one more thing about the mobilewallet shango script. It now looks up for the local ip, this should be the domain name or external ip if used to connect from outside. I had to change this afterreading qr from shango

hopitas, 2018-09-14:
I followed this guide and got Shango wallet working from external ip: [https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_68_shango.md](url)

The 97addMobileWalletShango.sh points many times to

`./.lnd`

dir when it should point to

`/mnt/hdd/lnd`

Fix these and the ip thing and I guess it should work :)

rootzoll, 2018-09-15:
See link in this tweet - there is advice how to get it to work.
https://twitter.com/RobEdb/status/1041052180743184385

Use for the next fix of script.

rootzoll, 2018-12-08:
Test also with latest ZAP: https://twitter.com/ottosuess/status/1069983204520722432
If done - make new issue for integration of ZAP URLs.

rootzoll, 2018-12-11:
Use also the info from this issue to improve the tutorial text on how to handle the process on different OS terminal types: https://github.com/rootzoll/raspiblitz/issues/193

rootzoll, 2018-12-21:
OK .. prepared mobile wallets and dynamicDNS for version 1.0 release.

closed 2018-12-21 by rootzoll