raspiblitz/raspiblitz #94: Delete existing wallet

Jantokil, 2018-09-13:
Hello!

I tried setting up the raspiblitz, but I ended the installing procedure to finish it later. I remember getting the passphrase just before I ended the installing procedure but I thought I would get a new one when resuming the wallet creation. Now when I try to install a new wallet it only gives me the option to restore an existing wallet or generating a new one. When I choose the latter option it gives me an error:

`!!! FAIL !!! SOMETHING WENT WRONG:
[lncli] unable to generate seed: rpc error: code = Unknown desc = wallet already exists
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

Press ENTER to retry ... or CTRL-c to EXIT`

Restoring isn't an option because I haven't set up any wallet yet. 

Help would be much appreciated.

Thanks!

rootzoll, 2018-09-14:
Maybe you need to clean the lnd directory. ssh fresh into the blitz, cancel on first menu and run following command: sudo rm -r /mnt/hdd/lnd

then close terminal and ssh fresh in and continue setup.

Jantokil, 2018-09-14:
Thanks for helping @rootzoll! I deleted the folder, but then I got an error saying the directory isn't being found. So I flashed the sd again and run the installer. And now it works!

closed 2018-09-17 by rootzoll