raspiblitz/raspiblitz #97: Required reindex after copying blockchain

adolfosrs, 2018-09-16:
I have copied my bitcoin folder from another HD using the raspiblitz script but when running `bitcoind` it was asking me for a `reindex` and shutting the daemon down right away.

So I have set it to run with the reindex flag but it seems that it will take a lot of time though.

Is this expected and is there anything I can do to skip that?

vmavromatis, 2018-09-16:
I'm kind of in the same position - not sure if we have the same error. I choose to do the reindex on my desktop (just opened bitcoin-qt and pointed it to the folder) and then I am planning to copy it back to the rpi. Not sure if it's the right thing or just my stupidity, maybe I will get asked to reindex it again. Let's find out! Good luck!

CarlosRoldanx, 2018-09-16:
Reindexing may take a while (even 2 weeks), even if is not required anymore since it is running LND v0.5-Beta.

Just be patient :)

rootzoll, 2018-09-16:
yes the raspiblitz needs a indexed blockchain. indexing on deskop is much faster - should be done before copy if possible. working on better downloads torrent/ftp for the future to speed this up for the future.

closed 2018-09-16 by rootzoll

RemoRitzmann, 2022-03-05:
While trying to reindex I found out that the user needs to be correct:
For me it worked with 
sudo -u bitcoin bitcoind -reindex-chainstate