raspiblitz/raspiblitz #98: LND not ready yet ... waiting another 60 seconds.

frandani82, 2018-09-16:
I just updated my Raspberry 3 from raspiblitz v 8.6 to 9.2. At the end of the installation I got the following error:
LND not ready yet ... waiting another 60 seconds.
If this takes too long (more then 10min total) --> CTRL+c and report Problem

After that I deleted the installation and tried it a second time with the same error. 
LND not ready yet ... waiting another 60 seconds.
If this takes too long (more then 10min total) --> CTRL+c and report Problem

Any idea what the problem could be?

rootzoll, 2018-09-16:
how did you do the update? cleanHDD and new sd card image like described here:
https://github.com/rootzoll/raspiblitz#update-to-a-new-sd-card-release

frandani82, 2018-09-17:
I built my own SD Card according this instruction: https://github.com/rootzoll/raspiblitz/blob/master/README.md#build-the-sd-card-image

- Get a fresh Rasbian RASPBIAN STRETCH WITH DESKTOP card image: DOWNLOAD
- Write image to a SD card: TUTORIAL
- Add a file called ssh to the root of the SD card when mounted to enable SSH login
- Start card in Raspi and login per SSH with ssh pi@[IP-OF-YOUR-RASPI] password is raspberry

Now you are ready to start the SD card build script - copy the following command into your terminal and execute:

wget https://raw.githubusercontent.com/rootzoll/raspiblitz/master/build.sdcard/raspbianStretchDesktop.sh && sudo bash raspbianStretchDesktop.sh

rootzoll, 2018-09-17:
Ah OK .. when you build your own sd card image you will probably have the new bitcoin 1.7.0rc already ... it seems that they changed the database format for the index and make an migration when bticoind is starting. The script was not aware of this yet and is going into a loop sysing that LND is not ready yet for some hours until the migration is done. It could even try to continue while migration is running.

Go to terminal and run ./XXupdateScripts.sh ... still working on it, but the latest scripts from today should at least be able to detect migration and wait for it.

rootzoll, 2018-09-17:
After migration there is another prob with lnd config ...
ep 17 15:27:14 RaspiBlitz lnd[11859]: /home/bitcoin/.lnd/lnd.conf:22: unknown option: bitcoind.rpchost

this is work inprogress - may take some time

frandani82, 2018-09-17:
ok, thanks a lot

vmavromatis, 2018-09-18:
same issue here with the latest raspiblitz (built yesterday) it loops forever, then I run `./XXupdateScripts.sh` and then goes back to looping `LND not ready yet ... waiting another 60 seconds.`. Any ideas?

vmavromatis, 2018-09-18:
> After migration there is another prob with lnd config ...
> ep 17 15:27:14 RaspiBlitz lnd[11859]: /home/bitcoin/.lnd/lnd.conf:22: unknown option: bitcoind.rpchost
> 
> this is work inprogress - may take some time

@rootzoll So I read your comment and I just commented this line out from my lnd.conf, restarted lnd with `sudo systemctl restart lnd` and re-run the script ./10blabla ! My LND is syncing now!

machasm1, 2018-09-26:
There is an error in the lnd.conf file.
you are missing the [Bitcoind] flag before these entries

bitcoin.node=bitcoind
bitcoind.rpchost=127.0.0.1
bitcoind.rpcuser=username
bitcoind.rpcpass=password
bitcoind.zmqpubrawblock=tcp://127.0.0.1:28332
bitcoind.zmqpubrawtx=tcp://127.0.0.1:28333

thats why you get bitcoind.rpcuser error
so the file should look like this

# lnd configuration

[Application Options]
debuglevel=info
maxpendingchannels=5
alias=nameofnode
color=#68F442

# RPC open to all connections on Port 10009
rpclisten=0.0.0.0:10009
# Domain, could use https://freedns.afraid.org
#tlsextradomain=lightning.yourhost.com

[Bitcoin]
bitcoin.active=1

# enable either testnet or mainnet
#bitcoin.testnet=1
bitcoin.mainnet=1

[Bitcoind]
bitcoin.node=bitcoind
bitcoind.rpchost=127.0.0.1
bitcoind.rpcuser=username
bitcoind.rpcpass=password
bitcoind.zmqpubrawblock=tcp://127.0.0.1:28332
bitcoind.zmqpubrawtx=tcp://127.0.0.1:28333

[autopilot]
autopilot.active=1
autopilot.maxchannels=5
autopilot.allocation=0.6

geco91, 2018-10-26:
just checked - rpclisten=0.0.0.0:10009 is part of the actual lnd.conf, seems to work fine this way.

@rootzoll recommend to close this issue.

closed 2018-10-26 by rootzoll