raspiblitz/raspiblitz #100: Alternate Ports for bitcoind and lnd

jagottsicher, 2018-09-18:
If you run several RapsiBlitz in the same network behind a NAT firewall you want to forward ports, respectively use port triggering to lead the right data to particular nodes. 

For that it was nice to have the option to change the ports bitcoind and lnd are listening to. 

Also I would like to see an option to add additional ports opened in the firewall (ufw) before the reboot. 

I could imagine both features in the main menu under "advanced" OR in  the setup scripts before setup of the nodes, respectively before the firewall is enabled.

rootzoll, 2018-09-21:
The port displayed should be the one one from lncli getinfo / uris. Changing the port over the UI should be a future option. I never changed it, so experience here. Why lnd is not picking up the changed port you may get an answere at the LND Dev Slack.

piezzo, 2018-10-02:
The reason is that the frontend is running the command as a different user (admin instead of bitcoin), thus grabbing different configuration (lncli appears to directly access the config file and does not ask the server).

I'll make a PR later.

CryptoCloaks, 2019-02-21:
We have found a solution to this, that hopefully works for everyone!

1) Login to your node and get to the console

2) Stop lnd
`sudo systemctl stop lnd`

3) We will then disable lnd
`sudo systemctl disable lnd`

4) Now we will edit the lnd.service file
`sudo nano /etc/systemd/system/lnd.service`

5) Once the file is open look for
`ExecStart=/usr/local/bin/lnd --externalip=${publicIP}`

When you have found that line, we will add the port we want to start using instead of the default 9735 to the end like so (In our case we used 9736)
`ExecStart=/usr/local/bin/lnd --externalip=${publicIP}:9736`

6) We will save the file
CTRL + O

7) We will exit the file
CTRL + X

8) We will enable the lnd service again
`sudo systemctl enable lnd`

9) Now we will shutdown and restart the RaspiBlitz to have our changes go into effect.
`sudo shutdown -r now`

10) Verify changes have been made with:
`lncli getinfo`

11) You should now see your uris has changed! SUCCESS!
[image]

rootzoll, 2019-02-21:
To setup this on a fresh raspiblitz - also I think in /mnt/hdd/lnd/lnd.conf the entry `listen=0.0.0.0:9736`is needed and a firewall port needs to be open on 9736:
`sudo ufw allow 9736 comment 'lightning mainnet'`
`sudo ufw --force enable`

CryptoCloaks, 2019-02-21:
I will edit the post above.

rootzoll, 2019-02-21:
I am adding this to 1.1 release  milestone ... as an idea for a script in `/home/admin/config.scripts/lnd.setport.sh` that simply takes the port number as parameter and executes the needed changes automatically :)

rootzoll, 2019-02-24:
OK - the master branch contains now a experimental script called: `lnd.setport.sh`
Will be part of v1.1 if you have v1.0 running, you can get that script with running command: `/home/admin/XXsyncScripts.sh` before the command below. 

So to change the LND port you now just go to the terminal (X Option in menu). And run this command: `/home/admin/config.scripts/lnd.setport.sh [PORTNUMBER]`

Closing issue - but test results are welcome. Everybody have fun.

closed 2019-02-24 by rootzoll

CommanderPoe, 2019-10-26:
Would this be the same process if I want to change the rpc 10009 and rest 8080 ports to connect with my wallet? 

I will quickly explain maybe any of you can shed some light here... Since I'm on IPv6 my router assigns me a different port for all 4 standards; 8333,9735,10009 & 8080. Since I'm running behind tor the bitcoind and lnd ports (1st and 2nd) don't care much, but the last 2 are much needed whenever I want to connect with my Zap wallet for example. Would it be enough just to add 2 more lines to the lnd.conf with the new ports assigned by the router?

rootzoll, 2019-11-16:
For changing the RPC & REST ports you need to edit the LND config manually .. see your options in this sample LND config: https://github.com/lightningnetwork/lnd/blob/master/sample-lnd.conf

TrezorHannes, 2021-12-29:
> 
Suggestion to this script: Since Hybrid mode (Clearnet and Tor in parallel) was introduced in 14.1, this script doesn't cater a port adjustment properly. Why? It checks whether it's running in Tor mode enabled, and if so, aborts. However, in Hybrid Mode, it's still justified to have Tor running AND requiring a port adjustment. 

/home/admin/config.scripts/lnd.setport.sh
```
# check if TOR is on
source /mnt/hdd/raspiblitz.conf
if [ "${runBehindTor}" = "on" ]; then
  echo "FAIL - portnumber cannot be changed if TOR is ON (not implemented)"
  exit 1
fi
```

would need to be adjusted to check for Tor AND `lnd.conf => tor.skip-proxy-for-clearnet-targets=true`, to fail only if former is TRUE and latter is false.

commit bca4dbfa (2018-10-08): Merge pull request #131 from piezzo/get_public_ip_from_cli

Get public ip from cli (fixes #100)