raspiblitz/raspiblitz #103: Bitcoin Core 0.17.0rc3 Doesn't Exist

normandmickey, 2018-09-19:
raspbianStretchDesktop.sh breaks here...
wget https://bitcoin.org/bin/bitcoin-core-0.17.0/test.rc3/bitcoin-0.17.0rc3-arm-linux-gnueabihf.tar.gz

rootzoll, 2018-09-21:
yeah with bitcoin dos bug old rc3 binaries got depublicated - no rc4 binary is out. I am testing and building new sd card image at the moment.

closed 2018-09-21 by rootzoll

normandmickey, 2018-09-21:
I saw that you updated the script and tried it out already. It seems to
have worked but has been doing this for hours "LND not ready yet ...
waiting another 60 seconds.
If this takes too long (more then 10min total) --> CTRL+c and report
Problem 1537521846".  I cancelled out of it and checked blockchaininfo and
that is fine. When I rebooted it comes back up to the same.  Should I just
let it run?

rootzoll, 2018-09-21:
new sd card image is up. new scripts have bitcoin 0.17.0 ... is you have older blockchain (that is still in the torrent) it needs to build new index db that is part of 0.17.0 ... this can take multiple hours. The ftp should have already new index prebuild. Updating the blockchains (torrent is also quite old) will do for next release. I will be on vacation until the end of month .. so some delay until my batteries are recharged for the next sprint.