raspiblitz/raspiblitz #105: RECEIVE does crash into permission problem

jagottsicher, 2018-09-20:
When trying to creatin an invoive I get:

```
******************************
Create Invoice / Payment Request
******************************

COMMAND LINE:
lncli --chain=bitcoin addinvoice 100000

RESULT:
./BBcreateInvoice.sh: line 63: ./.error.out: Permission denied
******************************
WIN
******************************

Give this Invoice/PaymentRequest to someone to pay it:

You can use 'lncli --chain=bitcoin lookupinvoice ' to check the payment.

Press ENTER to return to main menu.

```

I think It was an awesome feature if a QR-Code was shown!

rootzoll, 2018-10-20:
added chmod for error.out - needs testing.

geco91, 2018-10-26:
@rootzoll test worked. recommend ticket for closing.

rootzoll, 2018-11-04:
On the last hackday we intergrated the RTL web interface to manage your node and make fundings and payments. See progress on this ticket: https://github.com/rootzoll/raspiblitz/issues/134

In the long term this should replace the shell scripts .. creating invoices is not there yet, but when QR codes will surely be a part of it ... as they are already when funding chain wallet over RTL.

So basic bug is fixed and QR codes will then come with RTL - will close ticket for now.

closed 2018-11-04 by rootzoll

saubyk, 2018-12-05:
@rootzoll you can create invoices now with RTL. We can add QR code for the invoices in the next build.