raspiblitz/raspiblitz #108: Switching to TESTNET leads to failed verification

jagottsicher, 2018-09-21:
Swiched a completely  snyced and set up node to testnet and snced and setup there again. 

A try to connect to a faucet for testnet end up this way: 
```
` lncli connect 0270685ca81a8e4d4d01beec5781f4cc924684072ae52c507f8ebe9daf0caaab7b@159.203.125.125:9735
[lncli] rpc error: code = Unknown desc = verification failed: signature mismatch after caveat verification
`
```

rootzoll, 2018-09-21:
testnet3 was quite messed up over the last of weeks. I kind of ignored it ... but will try to repair support for version 1.0 next month. Until then I will be afk for vacation.

rootzoll, 2018-12-01:
TODO: Retesting switching to testnet.
TODO: Use/test new network.chain.sh config script
TODO: Add Dialogs to 00menu.sh - replace 95switchTestMain.sh

rootzoll, 2018-12-05:
@jagottsicher about the failed verification ERROR -> when using lncli on testnet the parameter `--network=testnet` is mandatory

see https://github.com/Stadicus/guides/issues/240#issuecomment-428923161

rootzoll, 2018-12-05:
OK - I adapted and tested (one testrun) all scripts to run with testnet again. Will be part of next sd card release.  closing ticket for now.

closed 2018-12-05 by rootzoll