raspiblitz/raspiblitz #112: Error code -28 after updating

Dolu89, 2018-09-22:
Hi,
I got this error after I updated my rpi with the latest image of raspiblitz. (raspiblitz-2018-09-22.img)
I have cleaned the HDD using ./XXcleanHDD.sh before update image.

```
$ ./10setupBlitz.sh

OK - bitcoind is running
Next step run Lightning

*** Checking bitcoin ***
OK - bitcoind is running

*** Test if chaninnetwork is ready ...
1537605388
result()
error(error code: -28
error message:
Verifying blocks...)
2018-09-22T08:32:29Z  block index          117187ms
2018-09-22T08:32:29Z Opening LevelDB in /home/bitcoin/.bitcoin/indexes/txindex
2018-09-22T08:32:30Z Opened LevelDB successfully
2018-09-22T08:32:30Z Using obfuscation key for /home/bitcoin/.bitcoin/indexes/txindex: 0000000000000000
2018-09-22T08:32:30Z Upgrading txindex database... [0%]
Waiting 1 minute and then trying again ...
```

Dolu89, 2018-09-22:
Ok I understand now. It progress, now it's 10%. Maybe "error" is not the good word at this point? I'm closing the issue anyway. Thanks

```
*** Test if chaninnetwork is ready ...
1537606977
result()
error(error code: -28
error message:
Verifying blocks...)
2018-09-22T08:32:29Z Opening LevelDB in /home/bitcoin/.bitcoin/indexes/txindex
2018-09-22T08:32:30Z Opened LevelDB successfully
2018-09-22T08:32:30Z Using obfuscation key for /home/bitcoin/.bitcoin/indexes/txindex: 0000000000000000
2018-09-22T08:32:30Z Upgrading txindex database... [0%]
2018-09-22T09:00:39Z Upgrading txindex database... [10%]
Waiting 1 minute and then trying again ...
```

closed 2018-09-22 by Dolu89

geco91, 2018-10-25:
The new sd card image (raspiblitz-2018-10-20.img.gz) uses the new torents that should now have the new indexes - to this waiting time for upgrading the txindex should be gone.