raspiblitz/raspiblitz #114: Problem with Cashout

svorob, 2018-09-23:
I was updating the image yesterday and followed to procedure as prescribed in README. Closed channels no problem, but had trouble moving funds out through the GUI. When prompted to enter amount, I left it blank to empty wallet fully, entered my address, and got back "OK. That worked :)", but then my balance never changed and `lncli listchaintxns` did not show it either (last txns were the ones to close channels). Tried several times and same thing happened every time. Since then I managed to send it from cli manually tweaking fees until in worked, final balance was 0 satoshis. 
I'm guessing BBcashoutWallet.sh doesn't catch the error somehow.

rootzoll, 2018-10-03:
Yes it looks like the cashout script needs update on error catching. Adding it to my list for the next release. thx.

rootzoll, 2018-12-05:
I tested and upadted the CASHOUT script. It work on my tests now and also uses multiple tries to get the fees right. Will be part of next sd card relase. I am closing the ticket for now.

closed 2018-12-05 by rootzoll