raspiblitz/raspiblitz #115: screen burning?

cedricwalter, 2018-09-23:
Did you notice any screen burning? my PI is  synchronizing since 2 days now. Is there a way to stop the screen and only make it react when touched?

Thanks

jagottsicher, 2018-09-28:
I also would like to have some "suspend mode", which toggles on/off the screen on demand.

rootzoll, 2018-10-03:
I havent seen a screen burning yet - but could be a long run problem.

If anybody likes to invest time into the control if touchscreen element of the LCD we could add some "screensaver waking up on touchscreen" or "toggle on/off on touch" feature - that would be great. 

But personally I would like to invest my dev time into the switch to a ePaper Display after version 1.0 anyway - like this one: https://www.waveshare.com/wiki/2.7inch_e-Paper_HAT_(B) No problems with screen burning or having a light emtting device anymore. Driving down energy consumption and finally some buttons for basic interaction directly with device.

jagottsicher, 2018-10-08:
e-paper solution is an good idea, but actually I prefer the touch screen.

Due to the fact that any touch feature works only under arbitrary condition in text mode I think one of the first issues should be an X-Server. You can run a console in graphic mode then after startup.

If you have an x-Server the needed steps to achieve touch features and screensavers are quite low.

also a web gui was easy to implement and software like ZAP desktop can be activated locally! I think that is the much more sustainable solution which offers by far more options for the raspiblitz (e.g. as a standalone POS for both smaller and bigger biz and also as an built-in piece of hardware for vending machines and others.

rootzoll, 2018-10-16:
The idea is to make both possible - run the RaspiBlitz with the LCD or an ePaper. To make the ePaper possible there will be some changes in the future - introducting an display.service or something like that. Once this is there - anybody is free to extend the LCD from no-touch/ASCII to a touchscreen/Graphics mode from there on.

I will close this ticket for now - feel free to reopen it later on.

closed 2018-10-16 by rootzoll

nivs, 2021-04-02:
The backlight brightness of Waveshare LCD screens [can be controlled via GPIO pin 18](https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(B)#Use_GPIO_to_control_backlight_brightness). If a "screen saver" function is added, it can be used to blacken the display and turn the backlight off.

openoms, 2021-04-02:
@nivs great, but did you test this?
from the link it looks like needs soldering (and not even clear where):
```
Solder a 0R resistor on this pad or solder directly to connect. 
```
Also I have no Waveshare screens, but interested.

nivs, 2021-04-02:
I tested it just now and it works perfectly. Soldering to enable is a very simple procedure.

https://user-images.githubusercontent.com/1798879/113455834-4acae600-9414-11eb-8e41-9b3a47c95a34.mov

openoms, 2021-04-02:
@nivs wow that is perfect! very cool video, thank you for posting! 
I can't say that we can make this a standard, but a menu option (at least triggered from the CLI) would be really easy.
Will get one of those screens.
@rootzoll check this out

rootzoll, 2021-04-15:
Yes @nivs this is very cool :) I also dont think such modification can be "standard" but its an interesting hack maybe to consider for an enhanced custom hardware version. I added the general topic to v1.8 milestone where we will refactor the WebUI & LCD/touchscreen UX to see how close we can go to a "screensaver".