raspiblitz/raspiblitz #117: Download Issue

FlyingMotion, 2018-09-24:
it were downloaded more than 231230404 byte, see attached picture. Now it stopped at 231230408 and don´t finish the process. As I restarted the ssh the process was finished and the test for the readyness of the chainnetwork started.
[image]
[image]

rootzoll, 2018-10-03:
Back from vacation - sorry for the delayed response.

For the latest release (which you are running) the torrent download script looks if there is a "complete flag" set in the rtorrent metadata file. In all my test cases this was a proper detection that the torrent download was complete. Not sure why this didnt worked in your case.

I could also break monitor loop when download size is OK with an added timeout - but would like to avoid the extra complexity. 

I leave this issue open until verison 1.0 release - if anybody reports on the same problem or I can reproduce i will add the extra break case.

rootzoll, 2018-10-26:
#155 is merged. closing issue.

closed 2018-10-26 by rootzoll