raspiblitz/raspiblitz #118: Error code -28 appeared while testing the readyness of the chainnetwork

FlyingMotion, 2018-09-24:
First I thought the process will struggle, but after a while I registered some progress as you can see in the attached pictures.

tomski174, 2018-09-26:
Same issue after updating to the latest release.  Please help.

jagottsicher, 2018-09-28:
as I know bitcoind is going through the transactions again. It takes aweful long.

Ciphergyst, 2018-09-30:
I'm basically having the same issue, I noticed that the rpc error -28 was bouncing up the LCD status screen and the ip:port was no longer reachable, but it appeared as if I could still communicate with the daemon, the channel list and balance was still there, so I figured a reboot was needed. After reboot the RPC error persisted but this time, balance & channels were no longer there. I figured it was time to update the SD with the updated build. I was unable to force close channels and remove lnd or channel balance temporarily, which is no big deal since the last two times I've upgraded the SD image the channels and balance persisted. Fast forward to the re initialization of the new image and the same as above is occurring except instead of upgrading the tx database script with percentage, mine occasionally says rewinding blocks with a decreasing progress=0999549 then it aborts the block rebuild, shuts down and starts all over again. What is also odd is that the local loop back:8332 says unable to connect, prior to the new sd build I thought it was port 8333? or is that inconsequential?

rootzoll, 2018-10-03:
Having this long phase of a -28 error waiting for the chain network to get ready is due to the update to the 0.17.0rc4 realease of bitcoind, which needs to update the txindex database to the new format. 

With updating the downloaded blockchain data from FTP and torrent (issue #127) this should not be needed anymore. 

@Ciphergyst that there were blocks rewinding, thats a bit strange. Maybe some corruption of data happend - not sure. Did it worked out in the end?

Ciphergyst, 2018-10-04:
@rootzoll Yeah, I thought it was odd as well. After a day and a half of seemingly no progress. I ran the init.sh script. it reformatted the HDD into FAT32 which was also very odd. So I unplugged it from the pi, switched it to my laptop and reformatted it back to Ext4 and tried again. So far no dice, attempting to re download the torrent/blockchain data for the last few days.

rootzoll, 2018-10-18:
OK. With the updating of the blockchain downloads with the new txindex this should ne good. Closing issue.

closed 2018-10-18 by rootzoll