raspiblitz/raspiblitz #124: Automatic wallet unlock under LND 0.5

SaschaMajewsky, 2018-10-01:
Currently the guides on creating scripts to auto unlock your LND and make the node truly passively syncing without overwatching it are deprecated and no longer reliable functioning under LND 0.5.

For example this guide is affected: https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_6A_auto-unlock.md

We should find a new way to implement auto unlock as this function is a crucial aspect for keep nodes up for a longer period of time.

rootzoll, 2018-10-03:
I will try to get this into version 1.0 .. but any help from the community how to fix this would be great.

frennkie, 2018-10-06:
I wrote a Python wrapper that can help here #130

rootzoll, 2018-10-08:
I merged the PR from frennkie - use at your own risk ;)

closed 2018-10-08 by rootzoll