raspiblitz/raspiblitz #127: Update Torrent File and FTP download

rootzoll, 2018-10-03:
Focus of last release was to change to a more stable torrent client and a better FTP server. Beside some edge case reports this looks good. Next step is now to update the data of FTP and Torrent with new index for Bitcoin 0.17.0 - to get rid of the long time consuming update index after download.

arcbtc, 2018-10-09:
I had no luck with the torrent so switched to ftp, however it has been hanging at 98% for past 12hrs?
[image]

arcbtc, 2018-10-10:
Still stuck 24hrs later? I'll try pulling the chain from somewhere else

Smiggel, 2018-10-13:
Torrent fails at 98%. Setup wants to continue, but everytime it stops as it says the torrent fails. I can keep the data for a continued download, but it does not help. It keeps being stuck in this loop. Trying now using FTP.

rootzoll, 2018-10-15:
Yeah it should check for 253000000 not 259000000 - thats too big. Do you use the lastest SD card image? raspiblitz-2018-09-22.img.gz?

arcbtc, 2018-10-15:
I think so. Will flash sd and report back

Smiggel, 2018-10-16:
Ok, update. This morning, after a night of downloading, the download continues after 253000000 and then stopped. I exited the download process and continued the setup. So far it looks good. The node downloaded the remaining blocks and it's now continuing.

JupJar, 2018-10-17:
using latest SD Card build
no luck with the torrent so switched to ftp, however ......

[image]

rootzoll, 2018-10-18:
OK. FTP and Torrent looks also good on my side now. I wrote a bit about the new torrent setup (base&update): https://github.com/rootzoll/raspiblitz/blob/master/background_downloadBlockchain.md

I set this issue to close for now.

closed 2018-10-18 by rootzoll