raspiblitz/raspiblitz #129: Use auto nat discover starting in lnd 0.5-beta

frennkie, 2018-10-06:
This the same issue as Stadicus/guides#249

rootzoll, 2018-10-16:
I followed your guide. Making new build and testing. Much thx.

rootzoll, 2018-10-18:
On my testing it did not work and the lnd made a shutdown getting into a loop of unlock, shutdown, unlock, shutdown - so this setting is not godd as a default ... will put scripts to old state for now.

Maybe add an option later to activate this.

frennkie, 2018-10-18:
Did you allow your Raspiblitz to create UPnP port forwardings..?

rootzoll, 2018-10-18:
@frennkie the ufw rule "allow local LAN SSDP for UPnP discovery" is active - is that what you mean?

frennkie, 2018-10-21:
> Moving issue to version 1.0 milestone to at least make it an option to activate.

Yes, makes sense.. also I forgot, that there is still an issue with the nat option: lightningnetwork/lnd#2014. 

LND will not re-create the port forwarding if - e.g. after a 24 hour DSL disconnect the port forwarding is no longer active on the router. I have been running the code in the PR for two weeks now and it's working fine. But I have to see how and when this can be merged into the official code.

rootzoll, 2018-11-04:
@frennkie can you report on any progress on this issue with lnd?

For version 1.0 want to remember to put "auto nat discover" as an option into the service section, but deactivated by default for now.

frennkie, 2018-11-05:
I have been occupied otherwise... From my point of view LND needs to be changed.. Not sure how fast such a change can be incorporated..

rootzoll, 2018-12-05:
With the next sd card release the auto-nat feature can be switched on in the servcie section to experiment with. Closing this ticket for now.

closed 2018-12-05 by rootzoll

frennkie, 2018-12-06:
Fine with me. 

As long as the two PRs for lnd are not merge it would make sense to add some context/additional information for people who think about enabling the option.

rootzoll, 2018-12-11:
If you write up two/three lines with additional info I can add it as an notice during the install process.